hzeller/rpi-rgb-led-matrix #1072: P10 16x32 outdoor image invert or not mapping correct

fabioharry, 2020-05-29:
@hzeller Thank you very much for the wonderful library!

I'm in trouble and I don't know what the cause is.

I have a P10 16x32 outdoor 1/4 scan, but without pin marking, when I run the tests it seems to be cut in half and inverted.

I tested all the multiplexing options but they didn't work.

Any idea what it is?

Defect in the LED module or mapping?

[image]

**Command:**
0 - some rotating square
./demo -D 0 --led-cols=32 --led-rows=16 --led-brightness=20 --led-multiplexing=9 --led-row-addr-type=4 --led-slowdown-gpio=4 --led-no-hardware-pulse 

**Output:**
[image]

**Command:**
1  - forward scrolling an image
`./demo -D 1 --led-cols=32 --led-rows=16 --led-brightness=20 --led-multiplexing=9 --led-row-addr-type=4 --led-slowdown-gpio=4 --led-no-hardware-pulse runtext.ppm `
**Output:**
[image]

**Command:**
9 - Volume bars
./demo -D 9 --led-cols=32 --led-rows=16 --led-brightness=20 --led-multiplexing=9 --led-row-addr-type=4 --led-slowdown-gpio=4 --led-no-hardware-pulse 

**Output:**
[image]

hzeller, 2020-05-29:
This might be one of the other `--led-multiplexing` modes maybe ?

fabioharry, 2020-05-29:
> This might be one of the other `--led-multiplexing` modes maybe ?

I tried from mode 1 to 16, but to no avail. (--led-multiplexing)

I noticed that they are dividing 8x8 LEDs, should I develop a new mapping?

There is no pin HUB75 marking on the board.

Command:
1 - forward scrolling an image
`./demo -D 1 --led-cols=32 --led-rows=16 --led-brightness=20 --led-multiplexing=9 --led-row-addr-type=4 --led-slowdown-gpio=4 --led-no-hardware-pulse runtext.ppm`

:)

[image]

hzeller, 2020-05-29:
If none of the existing led-multiplexing choices don't work, maybe you need to write one indeed.

I am not surprised that there is no pin-marking, some panels do, others don't.

fabioharry, 2020-05-29:
> If none of the existing led-multiplexing choices don't work, maybe you need to write one indeed.
> 
> I am not surprised that there is no pin-marking, some panels do, others don't.

This is bad news for me, I don't know how to program in C ++.

Any tips or documents on how to do to develop the new mapping?

Should I implement in [lib/multiplex-mappers.cc](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/multiplex-mappers.cc)?

hzeller, 2020-05-29:
Yes, that is the right locaiton.

The C++ stuff should be mostly copy-paste and modifying an existing examples.
The most interesting part for you is to figure out how the coordinates are mapped in your particular panel.

alealexa, 2020-06-23:
> This is bad news for me, I don't know how to program in C ++.
>
> Any tips or documents on how to do to develop the new mapping?
>
> Should I implement in [...]

Hello @fabioharry, did you found solution? Same issue with 16x32 matrix.

Alexey

alevecchio, 2021-06-10:
Does anyone find a definitive solution? If not, I'm coding a new multiplexer model.

closed 2025-11-27 by marcmerlin