hzeller/rpi-rgb-led-matrix #1074: P8 16*32 Matrix - Anyone had success?

deantaylor, 2020-06-03:
I'm having trouble connecting these panels (I'm using this through FPP).  Does anyone have any example configurations I can try? I'm a dabbler, rather than an expert, so a little lost. Thank you in advance. 

I tested the first channel and this lit up when naturally expecting full red on the first LED :

[image]

Identifying marks on the panels (?) :

[image]

Hans470, 2020-08-18:
Did you got it working? My P8 16x32 panels are working with ``--led-multiplexing=3``

deantaylor, 2020-08-18:
@Hans470 Thank you for giving me the motivation to try this again. i'm trying this through via FPP which gives several options. I've been away from this for some time so can't remember what this is all meaning. Can you perhaps share your settings if any of this looks familiar? Many thanks.
[image]

deantaylor, 2020-08-18:
Here i'm sending basic artnet of a solid line scanning - there seems to be an offset on each row. 
[image]

Hans470, 2020-08-19:
What is FPP? 
As I have seen you are also located in Berlin. I you want you can provide me one of your modules and I can test it at my raspi.