hzeller/rpi-rgb-led-matrix #1075: Function request if possible

rijotech, 2020-06-03:
Hi, Mr Zeller. 
I was wondering if possible to add images transitions option to led-image-viewer in the future 
 for example `sudo led-image-viewer -C -f -w5 -t8  *.gif -transition name`.
Also there is a way to auto play all mp4 videos in sequence from a folder?

Thanks for this amazing library, this is my latest project.

[image]

xuniuer, 2020-06-11:
I guess you could write a shell script to do that.
for gif in `ls *.gif`
do
  sudo led-image-viewer -C -f -w5 -t8 $gif -transition name # as you presented
  sleep 1 # Need to have a delay?
done

The above will play all gif files. If you want to have a loop, an outer loop could be added.

rijotech, 2020-06-19:
I already have a script but the problem is there is no option for a transition yet.

xuniuer, 2020-06-22:
Please refer to link: https://superuser.com/questions/556029/how-do-i-convert-a-video-to-gif-using-ffmpeg-with-reasonable-quality

closed 2020-09-13 by rijotech