hzeller/rpi-rgb-led-matrix #1076: Always the same error: cannot import name 'RGBMatrixOptions' from 'rgbmatrix'

petike779, 2020-06-04:
Hi all
Using 2x 16x32 panels in parallel connection as 32x32, Pi4B+ 4G
Using all the instructions , successfully working examples on folders:

-examples-api-use folder
-utils folder
On python folder I have a problem...
I am fighting with this error like most of beginners I suppose:
$ sudo ./simple-square.py
Traceback (most recent call last):
File "./simple-square.py", line 2, in
from samplebase import SampleBase
File "/home/pi/display16x32/rpi-rgb-led-matrix/bindings/python/samples/samplebase.py", line 7, in
from rgbmatrix import RGBMatrix, RGBMatrixOptions
ImportError: cannot import name 'RGBMatrixOptions' from 'rgbmatrix' (/usr/local/lib/python3.7/dist-packages/rgbmatrix/init.py)**

Steps done as in readme instructions:

$ sudo apt-get update && sudo apt-get install python3-dev python3-pillow -y
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'python3-pil' instead of 'python3-pillow'
python3-dev is already the newest version (3.7.3-1).
python3-pil is already the newest version (5.4.1-2+deb10u1).
0 upgraded, 0 newly installed, 0 to remove and 32 not upgraded.

$ make build-python PYTHON=$(which python3)
make -C ../../lib
make[1]: Entering directory '/home/pi/display16x32/rpi-rgb-led-matrix/lib'
make[1]: Leaving directory '/home/pi/display16x32/rpi-rgb-led-matrix/lib'
/usr/bin/python3 setup.py build --build-lib .
running build
running build_py
running build_ext

$ sudo make install-python PYTHON=$(which python3)
/usr/bin/python3 setup.py install
running install
running build
running build_py
running build_ext
running install_lib
running install_egg_info
Removing /usr/local/lib/python3.7/dist-packages/rgbmatrix-0.0.1.egg-info
Writing /usr/local/lib/python3.7/dist-packages/rgbmatrix-0.0.1.egg-info

And here I cannot do nothing... Tried also a clean reinstall with downloading again with git, doing all the steps again. I tried also with python 2.7 usage, the same error.
Reading other comments I realize that I am missing simple step...
Thanks in advance
Peter

petike779, 2020-06-07:
Hi everyone
I found my problem. Maybe solution will help others also fighting with this issue.
On the file /urs/local/lib/python3.7/dist-packages/rgbmatrix/__init_.py is missing one argument:

# -*- coding: utf-8 -*-
from __future__ import absolute_import
__version__ = "0.0.1"
__author__ = "Christoph Friedrich <christoph.friedrich@vonaffenfels.de>"
from .core import RGBMatrix, FrameCanvas

Because this file is write protected, first I had to change it to writable:
sudo chmod 666 __init_.py
With nano adding the missing argument:

# -*- coding: utf-8 -*-
from __future__ import absolute_import
__version__ = "0.0.1"
__author__ = "Christoph Friedrich <christoph.friedrich@vonaffenfels.de>"
from .core import RGBMatrix, FrameCanvas, RGBMatrixOptions

Save it.
Don't forget to change back permissions:
sudo chmod 644 __init_.py

Now all python scripts from samples folders are working. I don't know how disapeared this argument.
Cheers
Peter

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin