hzeller/rpi-rgb-led-matrix #1078: About EMI from LED matrix

xuniuer, 2020-06-09:
Hello,

This is about EMI issue of LED matrix. 

Recently we establish a 256*32 led matrix (P8, 16 panels, based on HUB80). When it works, we find that it affects VHF seriously. When deployed in airport, Air Channel (110MHz~130MHz) has large noise. Of course, after shutting down the matrix, the noise of two way radio disappears. 

Although the implementation is not based on the project completely, I want to ask if the problem always exists in this area or not.  Have any advice about EMI produced by matrix.

Thanks in advance.

hzeller, 2020-06-09:
If you are using any of these techniques in any project that is not simply experimenting you need to of course take care of an EMI emissions
You have very long cables, a large surface area and very high square-wave frequencies, so this is very important in that case.

In general, you want to use shielding (put all the cables including the controller board in a grounded metal box, which essentially means covering the back of the panel) and also make sure to cover put  ferrites around all cables leaving that enclosure.
You might even need a fine metal mesh in front of the LED panel.

You must check with an EMI engineer for these things.

closed 2020-06-09 by hzeller

xuniuer, 2020-06-09:
Thanks for quick reply. I will check points you mentioned.
Thanks again!!!

xuniuer, 2020-06-09:
Update：We re-arrange the wires at the first stage - use short cables, the radio interference to air channel is less largely.