hzeller/rpi-rgb-led-matrix #1079: Can work this software with temperature control and light control?

Robycop58, 2020-06-09:
I look that all the GPIO are used in this software so I cant use one I/O port for use one temperature controller or/and one light intensity controller. I like have on display with the temperature indicator and control the parameter --led-brightness=x and x depend of one light controller. Is this possible?
Thanks

Robycop58, 2020-06-11:
> 
> 
> It's certainly possible, it only depends on how many parallel chains of panels you want to use. If you only use one or two you will still have a lot of GPIO pins available.

Thank you a lot.
I am a beginner so can you please tell me the pins (for GPIO) that are free if I don't use the chain n.3 and the chain n.2?

hrocha1, 2020-06-12:
Go to the [wiring](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#alternative-hardware-mappings) page and click on the "Table: GPIO-pins for each hardware mapping". All pins with Chain 2/ a Chain 3/ should be available to use.

Robycop58, 2020-06-12:
> 
> 
> Go to the [wiring](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#alternative-hardware-mappings) page and click on the "Table: GPIO-pins for each hardware mapping". All pins with Chain 2/ a Chain 3/ should be available to use.

So, if I understand correctly, if I don't use Chain 3 I can use the GPIO with PIN 3, 5, 8, 36, 37 and 40. And more if I don't use Chain 2 too I have free the PIN 29, 31, 32, 33, 35 and 38.
Is right?

hrocha1, 2020-06-12:
That kinda depends on the hardware that you use. For regular mapping it's GPIO 12, 5, 6, 19, 13 and 20 for chain 2 and 14, 2, 3, 26, 16 and 21 for chain 3. You can also add your own mapping or change the current one in [hardware-mapping.c](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/hardware-mapping.c) (in case you need specific pins).

Robycop58, 2020-06-12:
> That kinda depends on the hardware that you use. For regular mapping it's GPIO 12, 5, 6, 19, 13 and 20 for chain 2 and 14, 2, 3, 26, 16 and 21 for chain 3. You can also add your own mapping or [...]

Fantastic, thanks a lot. This is what I need to know........ :-)

closed 2020-06-12 by Robycop58