hzeller/rpi-rgb-led-matrix #1081: HELP

Baztech1990, 2020-06-16:
Hey im trying everything and i have no idea what im doing wrong, im very new to all this and im following all the instructions or i hope i am, i even tried the adafruit instructions but im just  getting problems, not finding the commands ect, is there a "stupid" guide for very very beginners, im trying to run 64x64 led module with PI 3a+, wiring direct no hat, please help, thank you

Robycop58, 2020-06-18:
Hi, you have to explain better for make that we can understand your problems.
In any case, follow this simply instructions:
- I think you have the connector from led panel like in this page: https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#alternative-hardware-mappings
- connect the female led panel connector with wires to Raspberry like in previous link.
- if you have only one chain, connect only the pins (Raspberry side) 6,7,10,11,12,13,15,16,18,19,21,22,23,24,26. Attention, is easy to wrong..... (for use other GND, use pin 20 or 25)
- use the command line (python or c or video) with the flags:
	--led-gpio-mapping=regular
	--led-cols=64
	--led-rows=64
	if don't work use --led-panel-type=FM6126A or FM6127 (look the IC type on the led panel board)
	play with the following flags until the led panel work correctly (you have only to try the combinations because depend from your panel type):
		--led-multiplexing=X (X from 0 to 16)
		--led-row-addr-type=X (X from 0 to 4)

If you have more problems explain us.....

Baztech1990, 2020-06-18:
where do i do theis?

use the command line (Python or C or video) with the flags:
--led-gpio-mapping=regular
--led-cols=64
--led-rows=64
if don't work use --led-panel-type=FM6126A or FM6127 (look the IC type on the led panel board)
play with the following flags until the led panel work correctly (you have only to try the combinations because depend from your panel type):
--led-multiplexing=X (X from 0 to 16)
--led-row-addr-type=X (X from 0 to 4)

Baztech1990, 2020-06-18:
basically i want to know, what do i do in terminal and how do i go from terminal to python idle to run a matrix?

hzeller, 2020-06-18:
Follow the steps here: https://github.com/hzeller/rpi-rgb-led-matrix#lets-do-it

```
make -C examples-api-use
sudo examples-api-use/demo -D0
```

If you want to use Python, follow the steps in https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/bindings/python to build the Python binding. but first make sure that you have the basic demo working, as everything will get more complicated if you have not established that baseline.

Baztech1990, 2020-06-18:
make -C examples-api-use
sudo examples-api-use/demo -D0

If i type that it tells me "no such file or directory. stop"
im running straight from my Pi 3A+

Baztech1990, 2020-06-18:
what im trying to do is run a main video and then trigger 5 videos, all to display on a 128x64 matrix so all the help would be much appreciated

hzeller, 2020-06-18:
Viewing videos on the Pi has its challenges as it is borderline not fast enough to do that.

Instead of veering into unknown territory, why not just use the [existing video-viewer](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/utils#video-viewer) and script that ?

Baztech1990, 2020-06-19:
okay now im going to ask a dumb question, how do i script it? and how do i load the matrix into python3 idle and make it play on the module?

hzeller, 2020-06-19:
I don't know you have some idea to show one or more videos, I suspect you might just have a shell script that invokes the video-viewer with a bunch of these. You can also invoke the video viewer with multiple videos that it then just plays in sequence. So if you just want to play multiple videos, just pass them all on the command line.

I've never used Python, so I can't help you with that python question.

Baztech1990, 2020-06-22:
hi guys so i've been following: https://learn.adafruit.com/raspberry-pi-led-matrix-display/software and i do not get anything on my matrix and when i just run the program (sudo ./rpi-fb-matrix matrix.cfg) nothing happens and my pi freezes, any suggestions?  or diffrent way i can get my screen to output on my matrix?

hzeller, 2020-06-22:
If you're following some Adafruit tutorial you should probably ask Adafruit.

Baztech1990, 2020-07-06:
last question, how do i load the matrix into my code?