hzeller/rpi-rgb-led-matrix #1082: how get 32x32 rgb matrix to work properly

ccmic11, 2020-06-17:
Hi :)   I am new to C++ (am used to php) .   I am trying to get a 32x32 rgb led matrix to work to display text.   I am using it on Raspberry Pi 3 A+   

I first tried using the adafruit rgb matrix bonnet, and, after running the adafruit rgb matrix bonnet setup script (which uses -DDEFAULT_HARDWARE  adafruit-hat)   - all the Demos worked well, and the text-example did also - **ONLY** problem was that the top 16 pixels row did not display green at all, (even if i changed led-rgb-sequence parameter, it still displayed no green in top 16 pixels row).  (though Red and Blue worked fine)    (and the bottom 16 pixels row worked entirely fine, in all colors)

So, I then tried, without the adafruit bonnet,  following the instructions on here: "wiring up" (just using jumper leads) and ALL the colors worked OK, including green on all pixels.   However, it is glitchy:  The image below is WITHOUT using the adafruit bonnet: (led-gpio-mapping=regular)   

[image]

This was with slowdown=2   ,  (1, 3 or 4  were more glitchy).     I also tried adjusting a little bit with led-pwm-bits and nanosecond parameters but not much success. 

The led matrix is powered using a suggested 5v 3A barrel jack adapter.  

I do not mind whether the matrix works with OR without the adafruit bonnet - Either case will do -  I just would like to get it to work (with no glitches). 

Any ideas? 

thanks 

Mike

hzeller, 2020-06-17:
Sounds like your bonnet has a not properly connected green line. Follow the soldering that goes from pin 2 of the IDC connector to the HCT245 on the bonnet and the G1 line on the other side of that chip all the way to where it connects to the Pi.

(The glitchyness is because you connect it to the low-level IO pins on the Pi directly, you need the level-shifter in the bonnet)

ccmic11, 2020-06-18:
many thanks for advice -  the bonnet was pre-soldered, and the connections / soldering is so difficult to see and try follow the G1 connecions.  but yes, it seems perhaps that this bonnet might possibly be faulty at some point with its G1 soldering.