hzeller/rpi-rgb-led-matrix #1083: Is it possible to stop and resume the RGBmatrix updater thread?

stefandz, 2020-06-20:
I have a scenario where most of the time the LED matrix will be blank, but it can be worken up via a USB serial command to display something and the go back to sleep.

Is it possible to stop and later restart the RGBmatrix updater thread? I tried making it public so that I could access the Stop() and Run() methods, but I think I misunderstand the object model, as it seems that the updater thread is not a complete class at the point of declaration.

Any pointers anyone can offer would be greatly appreciated. The main motivation here is just to save CPU cycles and hence keep the system temperature down.

daveythacher, 2020-07-04:
The HUB75 is a multiplexed interface. For many panels there is no controller or offloading, so data is constantly sent over. Some panels have drivers which support hardware PWM which reduces data from software PWM. However this is generally used only with the memory mapped panels as I understand it. To my knowledge the memory mapped drivers are not supported by this project. For many of these drivers a clock source and row multiplexing is still required while the data is stored and PWM is generated internally.

Point being the updater thread is required to put the image on the panel. The code is broke out a little strangely, but it is really a GPIO bit bang interface. Part of the design is probably due to the nature of the GPIO mapping and access of the main program.

In summary unless you want the display to go completely black you need to avoid starving the updater thread. I have mentioned this is not an overly ideal solution for this very reason. A better solution would be to have a controller, which is dedicated. However the author of this did not wish to do that.

How large is your panel? Depending on the size you can use a MCU. Color representation ends up being a problem for HUB75 bandwidth, this project is around 11^3. (Its been a while) The data rate is variable but not near max if I remember correctly. Size in affects the CPU usage and reduces the FPS. Color representation further complicates this. Loading down the system will starve thread which will reduce FPS.

This code is designed to use the CPU, so one approach is switching to a memory approach of drawing a whole frame bit map into memory. (I am not sure this is what it does, but I did not study it too hard.) Here the IO/memory performance can become a liability for system performance. For some applications it will never matter, while it will for others.

MCUs end up having trouble with larger setups due to computation problems. (PWM generation is expensive in software and the HUB75 uses awkward bus.) Even if the controller has enough RAM it takes too much time to regenerate the memory image. The Pi's DMA does not support certain things. Moving to a SPI controller it would but it does not support GPIO in a super meaningful way. (Which is surprising and not surprising at the same time.)

At this point people usually switch over to FPGAs which can handle PWM and the bus just fine. It can do all of this in a single cycle, which should lower the power consumption and heat. However I have not looked into that just yet. Based on rough calculations it could be. 

If small enough you could use MCU, which may be cooler/lower power consumption that Pi. This library is really using an indirect notion at high clocks, which generates heat/uses more power. Again based on rough estimates. NEON ARM instructions can accelerate frame generation for certain Pi's, since that is used in ARM MPUs as I recall. However you still have the high clock/indirect problem.

Note many MCU's suffer from many of the problems in MPUs like the Pi. 32 bit ones which have high RAM have certain issues, but for a dedicated controller that is not likely to matter. (It has to do with MCUs being designed like MPUs, aka incorrectly.)

In summary super loops and bit banging are terrible. There are better ways. (They make cleaner software too.) Many ways actually. There is a number of things here which can drive up the thermal. This project limits the color representation to keep the CPU usage down I think. The relationship of color representation to computation is exponential. Linear for every thing else.

Embedded systems really have sunk. Most people just understand so little yet believe they understand so much. Linux is general purpose, but can be used in embedded systems. Linux software generally follows non-real time batch model, but can be used in real time systems. Basically you need to understand the pit falls.

Simple notions of inter process communication of "real time" systems is usually a joke. A simple hardware FIFO or emulated notion of it would fix that. These would be concerns if the HUB75 was different or a controller was used. System/memory/IO bus interfaces are usually a joke too. Super loops are used to avoid ISR latency which is created from related notions of clock/memory/IO wall and cost/margins/yield. 

People use parallel cores to emulate DMA or specific hardware which can make the problem worse. Notion here is parallelization. However certain pitfalls are not understood here either like burst transfers/overhead, interface types/limits, etc. Here is where the generalized nature really gets you too. 

If you understand the pitfalls you are okay, you can convert them into features. The PIC24CH solved a good number of issues. There are other approaches too, but it has some of these solved. Some it solves in one way which for others it supports multiple. Hardware is your friend, if you allow it to be. Was not the story here, so to the OP you may be screwed. At least you know some of the possible reasons why.

stefandz, 2020-07-05:
Hi. Thanks for your extended comment and ideas. I understand how the HUB75 interface works - however, putting aside all of the pros and cons of using a Raspberry Pi, another application processor with an OS, an MCU, an FPGA / CPLD, it should be possible to pause the data output thread and keep the display in a blank state, given that you can clock out zero for all of the pixels in the current frame (in fact, just doing so for the current row is sufficient, as that is the only part of the display being actively driven). You can then send no further data until required by ending the thread, hence reducing power consumption.

So far I just haven't worked out a structure to correctly expose this functionality due to the way that the classes involved are structured (being incomplete at definition) and was hoping that either someone else has done this or might be able to point me in the right direction to implement this feature and submit a pull request.

daveythacher, 2020-07-05:
If you are willing to blank the display you can just drive the output enable signal. You can also relay off the display, but this would require resetting the library. Simply putting zeros will not likely be as helpful as you think. You would have to rewrite the library for that. I tried to explain that.

If that power reduction is enough. I would assume deleting the object and manually driving the OE pin is also enough. This has overhead which makes it less ideal. However it is easy to do without changing the code. 

The current library starts the thread automatically during creation of the matrix object. This is not exposed via the canvas. The canvas is a glorified memory mapping, with integrated basic routines. (Fill, clear, drawCircle, etc.) 

The thread can be shutdown. (Not sure it is designed to support this exactly.) This code is a bit of a mess. That being said it is doable. Basically add a new method in matrix to shutdown the thread via its current method. The matrix method would need to access the IO for OE driving. You would need one for the reverse to restart it. Remember to call start not run. (It is really a broken pthread notion.) 

The real fun part is finding the logic in files with so many subclasses. Lot of ways to write poor code. (Messy source tree, huge logic in files, updating/defining logic from multiple classes, tons of subclasses, etc.) Code like this is just random.

You can follow ThreadedCanvasManipulator line 59 of rpi-rgb-led-matrix/include/threaded-canvas-manipulator.h. That is really a means to fork the canvas updates into its own thread. Language mappings could also be fun. The C notion converts the API from C++.  (It converts C to C++ then converts it back to C.) There is supposed to be bindings for other languages and I am not even going to bother looking at them.

It looks like led-matrix-c is the place for all the C mappings which create simple functions to wrap up the C++ logic. The structure is really a class pointer that gets cast. So you would modify the C++ RGBMatrix class and updating mappings.

UpdaterThread is private variable updater_ in led-matrix.h line 346. GPIO is accessible via getter line 244. Code for RGBMatrix is led-matrix c/h. IO operation example exists in line 478 of framebuffer.cc. 

Hardware mapping line of interest is line 31 of hardware-mapping.h. Hardware mapping is on line 115 of framebuffer-internal.h. Could be in FrameCanvas as private. The better answer for hardware mapping is iterating the matrix_hardware_mappings on line 24 of hardware_mapping.c. However you may need params (Options struct) and InitHardwareMapping scan which is mentioned in line 75 of framebuffer-internal.h and defined on 355 of framebuffer.cc. RGBMatrix constructor calls InitHardwareMapping on line 309 of led-matrix.cc. params_ is private on line 336 of led_matrix.h. Options is defined on line 59 of led-matrix.h with interesting part on line 68.

I get Deja-Vu constantly. I still dislike this library for a number of reasons. You can be right but completely wrong. Its not what you know that gets you into trouble, its what you know that just ain't so. Good luck and Take Care!

raulzanardo, 2023-12-01:
Have there been any updates regarding this issue?

nyooc, 2024-03-25:
> I have a scenario where most of the time the LED matrix will be blank

I have exactly the same scenario: besides the matrix display I installed a push button to the Pi, and upon a push (which would happen ca. 5 times a day), I want the matrix to operate for about 10 minutes.

First it was unexpected to see the CPU usage stay up at 75% even after clearing the display, and that I cannot properly close or deallocate the matrix object. After some research this is not suprising after all, and the documentation even tells about it in the [limitations](https://github.com/hzeller/rpi-rgb-led-matrix?tab=readme-ov-file#limitations) section:

> The system needs constant CPU to update the display. Using the DMA controller was considered but after extensive experiments dropped due to its slow speed..

Fortunately, it is not so hard to overcome this: just **use a separate process for the LED matrix handling, and another one for controlling the former**. I am using the Python 3 bindings on a Pi 3B+, and the following code stub seems to work well using the [multiprocessing](https://docs.python.org/3/library/multiprocessing.html) standard lib (aimed for controlling Python functions as separate processes), but I guess I could also use the more general [subprocess](https://docs.python.org/3/library/subprocess.html) lib (aimed for controlling any shell-executable process). Probably I lose some speed and response time but in my case it's not a problem.

```python
# matrix.py
import signal
import sys
import time
from typing import Optional

from rgbmatrix import RGBMatrix, FrameCanvas

# I put the matrix state into a module-wide variable so that functions can work on it
STATE: Optional[tuple[RGBMatrix, FrameCanvas]] = None

def init() -> tuple[RGBMatrix, FrameCanvas]:
    """Initialize the matrix and canvas"""
    matrix = RGBMatrix(options=...)
    return matrix, matrix.CreateFrameCanvas()

def loop(matrix: RGBMatrix, canvas: FrameCanvas) -> None:
    """Display loop: refresh the matrix"""
    while True:
        canvas.Clear()
        draw_something(canvas)
        canvas = matrix.SwapOnVSync(canvas)
        time.sleep(...)

def cleanup(signum: int, frame):
    """Catch termination signal, clear the matrix, and exit"""
    matrix, canvas = STATE
    canvas.Clear()
    matrix.SwapOnVSync(canvas)
    sys.exit(0)

def run() -> None:
    """Tell the process to do cleanup when terminating, initialize and run the loop"""
    global STATE
    STATE = init()
    signal.signal(signal.SIGINT, cleanup)
    signal.signal(signal.SIGTERM, cleanup)
    loop(*STATE)
```

```python
# main.py
from multiprocessing import Process
from matrix import run

# This code can be put into a loop checking a button, or something else...
if __name__ == "__main__":
    # Fire up the matrix
    process = Process(target=run)
    process.start()
    # Now the display loop is running, CPU usage at 75%
    
    wait_some_time_or_do_something()

    # Send a SIGTERM to the process, and it will execute the cleanup function
    process.terminate()
    # Now the display is blank, CPU usage down :-)
```

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin