hzeller/rpi-rgb-led-matrix #1084: artefacts on 64x64

alucardxs, 2020-06-21:
Hello,

It works pretty well with my RGB bonnet and my Pi zero but I've got some artefects on rows ... what is the cause ?

Thanks !

hzeller, 2020-06-21:
You don't write what you already looked at. In particular: did you do the necessary [PWM hardware modification](https://github.com/hzeller/rpi-rgb-led-matrix#improving-flicker) to avoid these situations ?

alucardxs, 2020-06-21:
Thanks for your answer.
I did not solder those two pins atm, I've got a pi Zero, are those the same ? (pin 4 to pin18) ?
Btw, can the LED panel be controled by android ? (putting images / animations directly like the pixelart does) ?

Thanks ;)

hzeller, 2020-06-21:
Yes Pi Zero pin output is the same. Also, the Bonnet might look slightly different than the Adafruit HAT, but the spirit of the modification is the same, same wiring.

You can easily write a network protocol which allows you to send pictures from your phone to the matrix. This is a library, that provides you a canvas object to set pixels on - all you need to set your imagination free and implement what you need.

You could use existing servers (such as [FlaschenTaschen](https://github.com/hzeller/flaschen-taschen) with the UDP  [Flaschen Taschen protocol](https://github.com/hzeller/flaschen-taschen/blob/master/doc/protocols.md#flaschen-taschen-protocol) and write a client in Android) or write your own which is of course more fun.

alucardxs, 2020-06-21:
Thanks dude for your help ;)