hzeller/rpi-rgb-led-matrix #1085: How to configure LED matrix size 32x16 to work as 64x32 using Adafruit RGB Matrix Bonnet

javiervalero31, 2020-06-22:
I've four LED matrix 32x16 connected to my RPi 3B through a [Bonnet](https://www.adafruit.com/product/3211)... 
I need to the matrices works in a resolution of 64x32 but I don't find a configuration to it works.
[image]

I want find a configuration using the C++ Demos to after put in my Python it code with `RGBMatrixOptions` and all else.

I have tryed the follow:

`sudo ./demo --led-rows=32 --led-cols=64 --led-slowdown-gpio=3 --led-multiplexing=0 --led-scan-mode=0 --led-row-addr-type=0 led-chain=1 led-parallel=1 -D 0`
[image]

`sudo ./demo --led-cols=64 --led-rows=32 --led-slowdown-gpio=1 --led-row-addr-type=0 -D 0`
[image]

`sudo ./demo --led-rows=16 --led-cols=128 --led-chain=1  --led-parallel=1 --led-pwm-dither-bits=1 --led-pwm-bits=7 --led-panel-type=FM6126A --led-row-addr-type=0 -D0`
[image]

`sudo ./demo --led-rows=32 --led-cols=128 --led-chain=1  --led-parallel=1 --led-pwm-dither-bits=1 --led-pwm-bits=7 --led-panel-type=FM6126A --led-row-addr-type=0 -D0`
[image]

`sudo ./demo --led-rows=32 --led-cols=128 --led-chain=1  --led-parallel=1 --led-pwm-dither-bits=1 --led-multiplexing=1 --led-pwm-bits=7 --led-panel-type=FM6126A --led-row-addr-type=0 -D0`
[image]

`sudo ./demo --led-rows=32 --led-cols=128 --led-chain=1  --led-parallel=1 --led-pwm-dither-bits=1 --led-multiplexing=2 --led-pwm-bits=7 --led-panel-type=FM6126A --led-row-addr-type=0 -D0`
[image]

Multiplexing, Rows conf, Columns conf and some more But I can not achieve the wished.
Maybe I am not seeing the error... Please Help!!!

hzeller, 2020-06-22:
First find the right settings that make your panel work ( `--led-cols=128 --led-chain=1 --led-parallel=1 --led-panel-type=FM6126A` above looks like the most promising.

Now you need to tell the library in which way you have arranged the layout to remap the coordinates to behave like a 64x32; check out the various ways to do that at

https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#remapping-coordinates

javiervalero31, 2020-06-23:
Yeah! Thanks,
I Ordered the matrices and connected it like this -->[image]
After, I used the flag --led-pixel-mapper=U-mapper and I got my configuration wished!
Now, I can display my opening GIF good...

Thank you very much !
[image]

closed 2025-11-27 by marcmerlin