hzeller/rpi-rgb-led-matrix #1086: Brand New RGB Adafruit 64x32 - only blue on top half

craigsca, 2020-06-22:
Working on a project and just started to put it together today.  A lot of troubleshooting, but it seems I finally have the display working - sort of.

Raspberry Pi 3 A+
64x32 RGB LED Matrix - 5mm pitch
5v power supply
RGB Matrix Bundle from Adafruit

For some reason, it seems I'm only get blue color on the top half, whereas the rest of the panel is show all colors.  I looked up other cases, one in which a person had an issue with blue up top, but no coloring on the bottom and it was a wiring issue.  However, I can't see how I would have that problem since I'm using the bonnet.

[image]
[image]

The clock should look like this: (https://raw.githubusercontent.com/riffnshred/image_bank/master/nhl-led-scoreboard/documentation/network_indicator.png)

hzeller, 2020-06-22:
As first step, I'd try to reduce the clocking speed (`--led-slowdown-gpio`), in case it is a speed issue with the particular shift register.

(You did set the `--led-gpio-mapping=adafruit-hat` right ?).

But other than that: it either is a wiring problem (e.g. soldering problem on the Bonnet ?) or an issue on the matrix board itself.

hzeller, 2020-06-22:
Of course, the usual suggestions in https://github.com/hzeller/rpi-rgb-led-matrix#troubleshooting apply, e.g. if there is something that messes with your GPIO pins in parallel, this can be a problem as well.

craigsca, 2020-06-22:
Yes, should have been more thorough...I am using --led-gpio-mapping=adafruit-hat and I've tried 0 through 3 in the slowdown-gpio settings.  

I looked at the troubleshooting guide and didn't see anything that described my problem.  The closest thing I could try was changing the RGB with --led-rgb-sequence but it didn't change the all-blue of the top 16 rows.

I will continue to try different things.  I wish there was a way to test the matrix itself or the bonnet.  

Thank you for replying so quickly.

hzeller, 2020-06-22:
You can also manually wiring it up: 
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md

(with the caveat, that this will not have level shifters)

craigsca, 2020-06-27:
Ordered a new matrix and a couple more bonnets.  I think I've found the issue - the hihat unfortunately does not have enough clearance on a Pi3 and it seems when I push the hihat completely "on" it comes in contact with the Pi3 USB connection.  When I pull the hihat away from physically touching, the screen works fine.  Go figure.  Just ordered a Pi4 because it looks like it will not have that problem.  

Thank you for the troubleshooting tips!

closed 2020-06-27 by craigsca