hzeller/rpi-rgb-led-matrix #1087: Help with scrolling 2 lines of text with runtext.py  or other *.py

absolutspl, 2020-06-23:
I have read through all the pertinent info from adafruit and rpi-rgb-led-matrix. Sorry if this is a repeated question.
Is it done with multiple lines? like this: 
self.parser.add_argument("-t", "--text", help="The text to scroll on the RGB LED panel", default="ZZZZZZ")
        self.parser.add_argument("-t", "--text", help="The text to scroll on the RGB LED panel", default="XXXXXX")
Thanks in advance.

hzeller, 2020-06-23:
Not clear what you mean, but do you want two different lines shown on the panel ?
Of course - just modify the code to do that. You have the power of Python or C++ at your hands, you can do anything you want with the canvas the LED matrix provides.

absolutspl, 2020-06-24:
I am a systems/network administrator. I am only lightly versed in python,
near zero  in C++.  I truly appreciate all of your coding and help to
everyone on the forum.

If I stick with runtext.py, can you point me  where to change the code.

*~/rpi-rgb-led-matrix/bindings/python/samples $ cat
runtext.py#!/usr/bin/env python# Display a runtext with
double-buffering.from samplebase import SampleBasefrom rgbmatrix import
graphicsimport timeclass RunText(SampleBase):    def __init__(self, *args,
**kwargs):        super(RunText, self).__init__(*args, **kwargs)
self.parser.add_argument("-t", "--text", help="The text to scroll on the
RGB LED panel", default="ZZZ")    def run(self):        offscreen_canvas =
self.matrix.CreateFrameCanvas()        font = graphics.Font()
font.LoadFont("../../../fonts/helvR12.bdf")        textColor =
graphics.Color(255, 255, 0)        pos = offscreen_canvas.width
my_text = self.args.text        while True:
offscreen_canvas.Clear()            len =
graphics.DrawText(offscreen_canvas, font, pos, 10, textColor, my_text)
      pos -= 1            if (pos + len < 0):                pos =
offscreen_canvas.width            time.sleep(0.05)
offscreen_canvas = self.matrix.SwapOnVSync(offscreen_canvas)# Main
functionif __name__ == "__main__":    run_text = RunText()    if (not
run_text.process()):        run_text.print_help()*

anandrajgupta, 2022-07-19:
I also have a similar requirement.

Please provide if a solution exists!

I am using c-code to make program for running two lines simultaneously, and for that i will have to run 2 for loops simultaneously inside main() function, is it possible?

i think for that i will have to create a thread and in that thread the "canvas" will not be recognized, the "canvas" is only accessible in main() function isn't it?

Please guide me in the right direction, Thank you!

absolutspl, 2022-07-19:
I will have to look over my notes and see how i finally got that to work. I
will get back to you shortly.

anandrajgupta, 2022-07-21:
Thank you very much, i will be waiting...

absolutspl, 2022-07-21:
Results with 2 lines of scrolling text that updates every 5 minutes
Location of temp sensor, time of last read from sensor, and temp &
humidity of last read.

:[image: image.png]

[image: image.png]

absolutspl, 2022-07-21:
With script & python i am :
1 taking temperature readings from a database
2 combining all the information into 1 line
3 adjusting color of led text based on temp or humidity
4 creating a ppm
5 displaying ppm

Here my information Dump:

**_pi@BondRoomLed:~ $ sudo crontab -l_**
# Edit this file to introduce tasks to be run by cron.
#
# Each task to run has to be defined through a single line
# indicating with different fields when the task will be run
# and what command to run for the task
#
# To define the time you can provide concrete values for
# minute (m), hour (h), day of month (dom), month (mon),
# and day of week (dow) or use '*' in these fields (for 'any').
#
# Notice that tasks will be started based on the cron's system
# daemon's notion of time and timezones.
#
# Output of the crontab jobs (including errors) is sent through
# email to the user the crontab file belongs to (unless redirected).
#
# For example, you can run a backup of all your user accounts
# at 5 a.m every week with:
# 0 5 * * 1 tar -zcf /var/backups/home.tgz /home/
#
# For more information see the manual pages of crontab(5) and cron(8)
#
# m h dom mon dow command
*/5 * * * * /home/pi/rpi-rgb-led-matrix-master/ledcron >> /home/pi/rpi-rgb-led-matrix-master/error.log 2>&1

**_pi@BondRoomLed:~ $ cat /home/pi/rpi-rgb-led-matrix-master/ledcron_**
#!/bin/bash
#A Bash script to copy data from database to prep led file,and display new data every 5 minutes by Shawn Logan
#/home/pi/rpi-rgb-led-matrix-master/ledcron#
#1 script started every 5 minutes from crontab
sudo /usr/bin/killall led-matrix
sudo /usr/bin/killall python
cd /home/pi/rpi-rgb-led-matrix-master/
sudo /home/pi/rpi-rgb-led-matrix-master/rem.sh

**_pi@BondRoomLed:~ $ cat /home/pi/rpi-rgb-led-matrix-master/rem.sh_**
#!/bin/bash
#A Bash script to copy data from database to prep led file, by Shawn Logan
#/home/pi/rpi-rgb-led-matrix-master/rem.sh
#2 Script
cd /home/pi/rpi-rgb-led-matrix-master/
echo " " >/home/pi/rpi-rgb-led-matrix-master/exp.txt
sudo mysql -sN -u root -p88888 -h 10.10.16.9 sensor_logs -e "SELECT DISTINCT device_desc FROM telemetry_log WHERE device_id='30' and unit_of_measure='F' ORDER BY date DESC Limit 1" >>/home/pi/rpi-rgb-led-matrix-master/exp.txt
sudo mysql -sN -u root -p88888 -h 10.10.16.9 sensor_logs -e "SELECT SUBSTRING_INDEX(date,' ',-1) FROM telemetry_log WHERE device_id='30' ORDER BY date DESC Limit 1" >>/home/pi/rpi-rgb-led-matrix-master/exp.txt
sudo mysql -sN -u root -p88888 -h 10.10.16.9 sensor_logs -e "SELECT DISTINCT Value,unit_of_measure FROM telemetry_log WHERE device_id='30' and unit_of_measure='F' ORDER BY date DESC Limit 1" >>/home/pi/rpi-rgb-led-matrix-master/exp.txt
sudo mysql -sN -u root -p8888 -h 10.10.16.9 sensor_logs -e "SELECT DISTINCT Value,unit_of_measure FROM telemetry_log WHERE device_id='30' and unit_of_measure='%' ORDER BY date DESC Limit 1" >>/home/pi/rpi-rgb-led-matrix-master/exp.txt
echo " " >>/home/pi/rpi-rgb-led-matrix-master/exp.txt
sudo mysql -sN -u root -p88888 -h 10.10.16.9 sensor_logs -e "SELECT DISTINCT device_desc FROM telemetry_log WHERE device_id='72' and unit_of_measure='F' ORDER BY date DESC Limit 1" >>/home/pi/rpi-rgb-led-matrix-master/exp.txt
sudo mysql -sN -u root -p 88888 -h 10.10.16.9 sensor_logs -e "SELECT SUBSTRING_INDEX(date,' ',-1) FROM telemetry_log WHERE device_id='72' ORDER BY date DESC Limit 1" >>/home/pi/rpi-rgb-led-matrix-master/exp.txt
sudo mysql -sN -u root -p 88888 -h 10.10.16.9 sensor_logs -e "SELECT DISTINCT Value,unit_of_measure FROM telemetry_log WHERE device_id='72' and unit_of_measure='F' ORDER BY date DESC Limit 1" >>/home/pi/rpi-rgb-led-matrix-master/exp.txt
sudo mysql -sN -u root -p 88888 -h 10.10.16.9 sensor_logs -e "SELECT DISTINCT Value,unit_of_measure FROM telemetry_log WHERE device_id='72' and unit_of_measure='%' ORDER BY date DESC Limit 1" >>/home/pi/rpi-rgb-led-matrix-master/exp.txt

sudo mysql -sN -u root -p 88888 -h 10.10.16.9 sensor_logs -e "SELECT DISTINCT Value FROM telemetry_log WHERE device_id='72' and unit_of_measure='%' ORDER BY date DESC Limit 1" >/home/pi/rpi-rgb-led-matrix-master/per.txt

echo $(cat /home/pi/rpi-rgb-led-matrix-master/exp.txt) > /home/pi/rpi-rgb-led-matrix-master/expNew.txt #Combine lines
sed "s/BondRoom/$(cat /home/pi/rpi-rgb-led-matrix-master/expNew.txt)/" /home/pi/rpi-rgb-led-matrix-master/Multi.py > /home/pi/rpi-rgb-led-matrix-master/Multi-line.py
sudo /home/pi/rpi-rgb-led-matrix-master/clr.sh

**_pi@BondRoomLed:~ $ cat /home/pi/rpi-rgb-led-matrix-master/clr.sh_**
#!/bin/bash
[... 23 lines cut ...]
fontsize = 14
font = ImageFont.truetype("/usr/share/fonts/truetype/freefont/FreeSans.ttf", fontsize)

# Determine text size using a scratch image.
img = Image.new("RGBA", (1,1))
draw = ImageDraw.Draw(img)
textsize = draw.textsize(text, font)

# Now that we know how big it should be, create
# the final image and put the text on it.
background = (0, 0, 0, 15)
img = Image.new("RGBA", textsize, background)
draw = ImageDraw.Draw(img)
draw.text((0, 0), text, fontcolor, font)

img.save("seesharp.ppm")

os.system("./led-matrix -t 288 -D 1 -c 2 seesharp.ppm")

Results with 2 lines of scrolling text that updates every 5 minutes
Location of temp sensor, time of last read from sensor, and temp &
humidity of last read.
[](https://user-images.githubusercontent.com/67323849/180222126-b425094b-a8af-49ff-9ebd-94a549ee0ea7.png)

[](https://user-images.githubusercontent.com/67323849/180222188-42a08cf6-972e-45ca-a6e7-0552a411d960.png)

anandrajgupta, 2022-07-22:
Thank you!

As i mentioned earlier i am using C-Code so this help is not usable for me, thank you for the efforts anyway :)

absolutspl, 2022-10-11:
In python i am :
1 taking temperature readings from a database
2 combining all the information into 1 line
3 adjusting color of led text based on temp or humidity
4 creating a ppm
5 displaying ppm

Here my information Dump:

***@***.***:~ $ sudo crontab -l*
# Edit this file to introduce tasks to be run by cron.
#
# Each task to run has to be defined through a single line
# indicating with different fields when the task will be run
# and what command to run for the task
#
# To define the time you can provide concrete values for
# minute (m), hour (h), day of month (dom), month (mon),
# and day of week (dow) or use '*' in these fields (for 'any').
#
# Notice that tasks will be started based on the cron's system
# daemon's notion of time and timezones.
#
# Output of the crontab jobs (including errors) is sent through
# email to the user the crontab file belongs to (unless redirected).
#
# For example, you can run a backup of all your user accounts
# at 5 a.m every week with:
# 0 5 * * 1 tar -zcf /var/backups/home.tgz /home/
#
# For more information see the manual pages of crontab(5) and cron(8)
#
# m h  dom mon dow   command
*/5 * * * * /home/pi/rpi-rgb-led-matrix-master/ledcron >>
/home/pi/rpi-rgb-led-matrix-master/error.log 2>&1

***@***.***:~ $ cat /home/pi/rpi-rgb-led-matrix-master/ledcron*
#!/bin/bash
#A Bash script to copy data from database to prep led file,and display new
data every 5 minutes  by Shawn Logan
#/home/pi/rpi-rgb-led-matrix-master/ledcron#
#1 script started every 5 minutes from crontab
sudo /usr/bin/killall led-matrix
sudo /usr/bin/killall python
cd /home/pi/rpi-rgb-led-matrix-master/
sudo /home/pi/rpi-rgb-led-matrix-master/rem.sh
#sudo python /home/pi/rpi-rgb-led-matrix-master/Multi-clr.py

***@***.***:~ $ cat  /home/pi/rpi-rgb-led-matrix-master/rem.sh*
#!/bin/bash
#A Bash script to copy data from database to prep led file, by Shawn Logan
#/home/pi/rpi-rgb-led-matrix-master/rem.sh
#2 Script
cd /home/pi/rpi-rgb-led-matrix-master/
echo " " >/home/pi/rpi-rgb-led-matrix-master/exp.txt
sudo mysql -sN -u root -p88888 -h 10.10.16.9 sensor_logs -e "SELECT
DISTINCT device_desc FROM telemetry_log WHERE device_id='30' and
unit_of_measure='F' ORDER BY date DESC Limit 1"
>>/home/pi/rpi-rgb-led-matrix-master/exp.txt
sudo mysql -sN -u root -p88888 -h 10.10.16.9 sensor_logs -e "SELECT
SUBSTRING_INDEX(date,' ',-1) FROM telemetry_log WHERE device_id='30' ORDER
BY date DESC Limit 1" >>/home/pi/rpi-rgb-led-matrix-master/exp.txt
sudo mysql -sN -u root -p88888 -h 10.10.16.9 sensor_logs -e "SELECT
DISTINCT Value,unit_of_measure FROM telemetry_log WHERE device_id='30' and
unit_of_measure='F' ORDER BY date DESC Limit 1"
>>/home/pi/rpi-rgb-led-matrix-master/exp.txt
sudo mysql -sN -u root -p8888 -h 10.10.16.9 sensor_logs -e "SELECT DISTINCT
Value,unit_of_measure FROM telemetry_log WHERE device_id='30' and
unit_of_measure='%' ORDER BY date DESC Limit 1"
>>/home/pi/rpi-rgb-led-matrix-master/exp.txt
[... 56 lines cut ...]
51.87 %"
text = textwrap.fill(userinput, 38)

# Font size, color and type.
fontcolor = (0,255,0)
fontsize = 14
font =
ImageFont.truetype("/usr/share/fonts/truetype/freefont/FreeSans.ttf",
fontsize)

# Determine text size using a scratch image.
img = Image.new("RGBA", (1,1))
draw = ImageDraw.Draw(img)
textsize = draw.textsize(text, font)

# Now that we know how big it should be, create
# the final image and put the text on it.
background = (0, 0, 0, 15)
img = Image.new("RGBA", textsize, background)
draw = ImageDraw.Draw(img)
draw.text((0, 0), text, fontcolor, font)

img.save("seesharp.ppm")

os.system("./led-matrix -t 288 -D 1 -c 2 seesharp.ppm")

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828