hzeller/rpi-rgb-led-matrix #1088: Text Example in Python

ElBertrando, 2020-06-23:
Is displaying clock time in Python as easy as modifying the runtext.py example to input time.clock() as an input parameter ? Do I need to consider anything else in this scenario ? 
In this case I can just pass time.clock() as input parameter "textToDisplay" in this function:

```
    def runTextWithSpeed(self, speedOfText, textToDisplay, textColorToDisplay, displaySomething):
        if(displaySomething):
            offscreen_canvas = self.matrix.CreateFrameCanvas()
            font = graphics.Font()
            font.LoadFont("../../../fonts/7x13.bdf")
            #textColor = graphics.Color(255, 255, 0)
            textColor = textColorToDisplay
            pos = offscreen_canvas.width
            thisText = textToDisplay

            if(speedOfText == 0):
                pos = 20
                offscreen_canvas.Clear()
                graphics.DrawText(offscreen_canvas, font, pos, 10, textColor, thisText)
            elif(speedOfText > 0):
                while True:
                    offscreen_canvas.Clear()
                    len = graphics.DrawText(offscreen_canvas, font, pos, 10, textColor, thisText)
                    pos -= 1
                    if (pos + len < 0):
                        pos = offscreen_canvas.width
                    time.sleep(1.0/speedOfText)
                    offscreen_canvas = self.matrix.SwapOnVSync(offscreen_canvas)
        else:
            offscreen_canvas.Clear()
```

Thanks

UCSF-ETS, 2020-07-14:
we can close this as I am working on this clock solution

ElBertrando, 2020-07-14:
we can close this as I am working on this clock solution

closed 2020-07-14 by hzeller