hzeller/rpi-rgb-led-matrix #1089: Pi-3/Python getting hot question

bobblkabb, 2020-06-24:
Question of understanding
I control a **64x64 matrix** with **Python** on a **Pi-3**.
I have built a clock (DrawText, SwapOnVSync) and the hardware gets **'critically' warm** after a short time (> 70°C). It seems that brightness plays a role here. Yesterday I turned it down to 10 and I 'only' got temperatures around 60°C. But maybe I did not let it run long enough. 
The documentation says that C++ code is about 8x faster than Python code. Can you say that the temperature of my app in C++ is also much cooler than with Python?
Or is there any technical jumbo mumbo I don't know that says otherwise?

hzeller, 2020-06-24:
CPU usage is a problem, brightness should not make a difference.
The LED matrix requires one core that constantly runs to populate the matrix, so you can expect one of the cores to run at 100% all the time.
Calling a C++ function through the Python interface is indeed pretty slow if you set pixel by pixel for instance. Other functions like DrawText() or other graphics functions are more higher-level, i.e. there is less data transferred between Python code and C++ code, so the speed impact is much less.
If you're mostly just drawing text and SwapOnVSync() there should not really be too much impact due to Python.

If you do a lot other things in Python, then it can easily use more CPU cores - look with `top` how much CPU percent you're using.
(You can try the [clock example written in C++ if you want to compare](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use) ).

Problem with the Pi of course that the processor is covered up by the bonnet, so this might reduce air circulation. I guess making sure that there is enough ventilation, not mounting directly next to a hot power supply, maybe adding a heatsink, not overclocking etc. are probably good starting points addressing it. I never had temperature issues with a Pi, so I can't really be helpful here.

bobblkabb, 2020-06-24:
Many thanks for the quick answer!
The core utilization is approx. 60/70%.

Good idea to compare with the example program, I will do. 

I also thought about covering the processor and ordered an extension cable for the GPIO interface. Just hasn't arrived yet.

bobblkabb, 2020-06-26:
Cable solved the problem. Around 50°C (122°F) now.

closed 2020-06-26 by bobblkabb