hzeller/rpi-rgb-led-matrix #1090: Blue LED's Not working

orion-achillios, 2020-06-27:
I have recently begun a summer project with a friend using this git repository to control the Adafruit RGB LED matrix with the Adafruit RBG LED matrix hat with the real-time clock. When we first put everything together we had issues getting the LED's to register the color blue after running a few of the demos. At first, we thought it was due to bad soldering but after not finding any errors with our soldering with the first Adafruit hat we decided to try again with a new Adafruit hat and we are still getting the same error. I'm pretty good with C++ so I wrote a code that allows the user to input an RGB color sequence and has that color fill the matrix screen for testing purposes. The only colors not showing properly are colors that have any blue in them. Is it possible that the RBG LED matrix display I have purchased is broken? Please help also please use somewhat understanding terms because I'm in high school.

hzeller, 2020-06-27:
"_Is it possible that the RBG LED matrix display I have purchased is broken_?"

Possibly.

Though typically this often only manifests in the upper half or lower half of the board.
Also there are some panels out there that only support red and green, but these are typically HUB12-style panels.

If there is a printing on the back near the input connector, check out if that is looks a bit like in the [wiring page](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#connection).

orion-achillios, 2020-07-06:
the print on the back of the panel looks the same as the print you have on the wiring page. 

I wired it all by hand instead of using the hat and now I am seeing flashes of blue at the start and end of some of the demos but it's not actually displaying when needed. What should I do?

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin