hzeller/rpi-rgb-led-matrix #1091: Mapping Problem 64x64 "U-mapper"

Mylan101, 2020-06-27:
Hey, I have got four 32x32 matrixes connected in an "u-shape" (see image below).
[image]
Now when  I use the "U-mapper" like this:
[image]
I end up with two swapped halves looking like this:
[image]
I tried using the input on the opposite side also resulting in swapped halves just the other way around then.
Any help would be much appreciated.
Thanks in advance.

Mylan101, 2020-06-28:
I guess I now know why I have this problem, but I'm still unclear about how to fix it.
The way the panels are connected is still the same way it is described 

> So the following chain (Viewed looking at the LED-side of the panels)
    [<][<][<][<] }- Raspberry Pi connector
But instead of this configuration:
> is arranged in this U-shape (on its side)
    [<][<] }----- Raspberry Pi connector
    [>][>]
Mine looks more like this:
`                                             [<][<]
Raspberry Pi connector-----{[>][>]`

This explains why the halves are swapped because it is just folded the wrong way around for the "U-mapper".
The problem now becomes that I can't physically change the layout to the way the "U-mapper" was made for.
The connector cable connecting the rows would have to be more than twice as long then.
Did anyone here have a similar issue?
At this point, I'm thinking about just making a longer cable myself hoping that the signal integrity will still be fine.

Mylan101, 2020-06-28:
The formatting didn't turn out the way I expected sorry guys.
I'll try that part again.
Mine looks like this:
```
                                              [<][<]
Raspberry Pi connector-----{[>][>]
```

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828