hzeller/rpi-rgb-led-matrix #1095: sparkfun 32x32 matrix does not turn up

nicolaveritas, 2020-07-01:
I'm using this matrix https://www.sparkfun.com/products/14646

I'm wiring directly from the panel to a raspberry 3 B+ with no breakout board 

the matrix is powered by a 5V 2A power supply 

I'm trying to run the example `demo D0` but the panel doesn't turn up at all. I've tried to tweak with all the different flags with no success (`--led-multiplexing`, `--led-row-addr-type`, `--led-panel-type`)

Is there any way to debug the code? What should I check to make sure everything is set up correctly?

hzeller, 2020-07-01:
did you follow the wiring guide ? https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md

nicolaveritas, 2020-07-01:
yes, I've did it multiple times and double check it. I also have 3 of these matrices and tried with all of them, even trying the connector labeled as output but I have the same result

when I turn on the power without being connected to the pi some leds flickers, if I do that when I'm connected that doesn't happens

hzeller, 2020-07-01:
You also did connect the gnd of the Matrix with the gnd of the Pi ?
If nothing turns on in that case, and you use have the proper hardware mapping enabled, then you probably need the digital level shifter.

nicolaveritas, 2020-07-01:
when you ask about GND you mean as explained in the wiring guide? if so yes, I have GND from the panel connected to pin 6 of the pi

As far for the hardware mapping I tried both with no args, which I assume defaults to regular, and setting regular explicitly (given that my wiring follows the guide https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md)

I have to admit I don't know how to use a digital level shifter, how should this fit in the wiring of this setup?

hzeller, 2020-07-01:
The level shifter would be easiest by building one of these
  https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3
or acquiring some of other providers https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/

nicolaveritas, 2020-08-02:
I have built this adapter https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3

but unfortunately the behaviour is the same, I have even tried with the same type of matrix from a different supplier

I'm running out of ideas so if anyone has any idea about what I should check again I'm very open to suggestion

nicolaveritas, 2020-08-02:
I managed to make it work building the adapter active-3

I had some troubles even with the adapters: at the beginning it looked like the behaviour was the same but the problem was just that I wasn't setting the param `--led-parallel=1`

thanks again for your work and the support

closed 2020-08-02 by nicolaveritas