hzeller/rpi-rgb-led-matrix #1096: Matrix was working, and now is showing flickering lines

michaelkennyjr, 2020-07-01:
I'm using a Raspberry Pi 3 B+ with an Adafruit HAT to run a single 64x32 HUB 75 matrix. I did the assembly at my dad's, and when I powered everything up, I could only get flickering lines (mostly red, some green) on the screen. After reading about similar problems, I thought I might be able to fix this with the --led-gpio-slowdown flag, but I had no luck.

I took everything home and plugged it in, figuring I was in for a long day of research and troubleshooting, but to my surprise the matrix worked perfectly, with all the right colors and minimal flickering. (--led-gpio-slowdown=2 seemed to work best.) Since then I've been able to tinker with my Python code to generate text and graphics and get everything to render well.

But this morning, I moved the Pi and the matrix to a different room in the house, plugged them in, and suddenly the matrix is back to how it was before: red and green flickering lines. No matter how I set the --led-gpio-slowdown, nothing seems to work. (Incidentally, setting the value to 5, which should be out of range, makes the images show up but in all red.) I went back to the power outlet where it had been working, and it's having the same issue. Everything was working fine, and now it's not, and I have no idea why.

I've included images of the Pi and matrix with wiring. The matrix is being powered through the HAT with a 5V/10A power supply.

[image]
[image]
[image]

davemaster, 2020-07-02:
Try with another power supply, 5V with 5A minimun

hzeller, 2020-07-02:
Check that GND is connected properly
Since you're using the adafruit hat, make sure to supply the right `--led-gpio-mapping`
Make sure that there is no other program running on your Pi fighting for access to GPIO (also check https://github.com/hzeller/rpi-rgb-led-matrix#troubleshooting )

michaelkennyjr, 2020-07-02:
I've suspected that the power supply might be an issue. I dug up a 4A power supply just to very briefly test the matrix, and it looked like it worked. I'm ordering a new one with proper current output and I will update when it gets here.

--led-gpio-mapping is set to "adafruit-hat" and all of the connections are fine as far as I can tell.

michaelkennyjr, 2020-07-06:
My new power supply came, and it's working great. It looks like that was the issue. Thank you!

closed 2020-07-06 by michaelkennyjr