hzeller/rpi-rgb-led-matrix #1098: Samplebase error: Need Root -- Python API

ElBertrando, 2020-07-06:
Thanks for your direction in pointing out that I wasn't calling `process()` that creates the matrix in the inherited class `samplebase.py `in my code for running text.  I am testing the code from Thonny IDE from the raspberry Pi 4. I created a folder "code" which contains my custom API code, and given it root access `chmod -R 777 ./`  
However I am still getting this root error when line 66 is called in samplebase.py when I debug:                                             `self.matrix = RGBMatrix(options = options)`
Pic here:
[image]

I edited the edit the `../../lib/Makefile ` to reflect my panel as well [adafruit-hat-pwm] ...

hzeller, 2020-07-06:
Once you read the error message which explains exactly what to do - What did you do then ?

ElBertrando, 2020-07-07:
Indeed I am trying to run the code using sudo from Terminal - I have tried "`sudo python3 /home/pi/MediasiteSignFbAPI/Code/Main.py`"  as well  "`sudo /home/pi/MediasiteSignFbAPI/Code/Main.py`" -- because the function call 
```
   run_text = RunT.RunText()
   run_text.process()
```

are called from Main.py.

ElBertrando, 2020-07-09:
Hi Henner,
I believe I solved this problem as well as the problem with my dependencies. My version of python was corrupted so I re-imaged and started again. 

 My next problem is that I am receiving AttributeError: 'RunText' object has no attribute 'matrix' when trying to display to the rgb matrix. 

[image]

Here is my entire class that I am calling to either Display the clock, or display text based on what I am getting from the API I am calling -- It is based very much on the runtext.py code in the python samples:

```
#!/usr/bin/env python

# Display a runtext with double-buffering.

from samplebase import SampleBase

from rgbmatrix import graphics

import time

# displayText = False

# displayClock = False

class RunText(SampleBase):

    displayText = False

    displayClock = False

    def __init__(self, *args, **kwargs):

        super(RunText, self).__init__(*args, **kwargs)

        self.parser.add_argument("-t", "--text", help="The text to scroll on the RGB LED panel", default="Hello world!")

    def run(self,bool,my_text, textColor):

        offscreen_canvas = self.matrix.CreateFrameCanvas()

        font = graphics.Font()

        font.LoadFont("../../../fonts/7x13.bdf")

        #textColor = graphics.Color(255, 255, 0)

        pos = offscreen_canvas.width

        #my_text = self.args.text

        if (bool == True):

            offscreen_canvas.Clear()

            len = graphics.DrawText(offscreen_canvas, font, pos, 10, textColor, my_text)

            pos -= 1

            if (pos + len < 0):

                pos = offscreen_canvas.width

            time.sleep(0.05)

            offscreen_canvas = self.matrix.SwapOnVSync(offscreen_canvas)

    def runTextWithSpeed(self, speedOfText, textToDisplay, textColorToDisplay):

        global displayText

        global displayClock

        displayClock = False

        displayText = True

        if(displayText):

            offscreen_canvas = self.matrix.CreateFrameCanvas()

            font = graphics.Font()

            font.LoadFont("../../../fonts/7x13.bdf")

            #textColor = graphics.Color(255, 255, 0)

            textColor = graphics.Color(textColorToDisplay)

            pos = offscreen_canvas.width

            thisText = textToDisplay

            if(speedOfText == 0):

                pos = 20

                offscreen_canvas.Clear()

                graphics.DrawText(offscreen_canvas, font, pos, 10, textColor, thisText)

            elif(speedOfText > 0):

                while True:

                    offscreen_canvas.Clear()

                    len = graphics.DrawText(offscreen_canvas, font, pos, 10, textColor, thisText)

                    pos -= 1

                    if (pos + len < 0):

                        pos = offscreen_canvas.width

                    time.sleep(1.0/speedOfText)

                    offscreen_canvas = self.matrix.SwapOnVSync(offscreen_canvas)

        else:

            offscreen_canvas.Clear()

    def runClock(self):

        global displayText

        global displayClock

        displayClock = False

        displayText = True

        offscreen_canvas = self.matrix.CreateFrameCanvas()

        font = graphics.Font()

        font.LoadFont("../../../fonts/7x13.bdf")

        textColor = graphics.Color(255, 255, 253)

        pos = offscreen_canvas.width

        while(displayClock == True):

            displayTime = time.strftime("%H:%M")

            offscreen_canvas.clear()

            graphics.DrawText(offscreen_canvas, font, pos, 10, textColor, displayTime)

            time.sleep(0.8)

            offscreen_canvas = self.matrix.SwapOnVSync(offscreen_canvas)
```

UCSF-ETS, 2020-07-14:
We can close this -- I found I was not recompiling the python bindings code after re-downloading the rgb matrix code

ElBertrando, 2020-07-14:
We can close this -- I found I needed to recompile my python while in that directory after re-downloading the code

closed 2020-07-14 by hzeller