hzeller/rpi-rgb-led-matrix #1100: 64x32 panel with amperka (adafrult)

anp135, 2020-07-13:
Hi. I have https://amperka.ru/product/rgb-led-matrix-driver-cap with some RGB matrix from Ali:
P4-2121-16S-HL 1.0 2P2751A 32x32
P4 256*128-2121A5 64x32
P5(2121)-16S-6432-B0 2PS1111F-HL 63x32
P6-3528-32*32-16S-HL1.1 2P2768A-HL 32x32

They are all upper half does not have blue, lower half does not have red. In the sabe time, when I use 
#./clock --led-cols=32 --led-rows=32 -d %H:%M:%S -f ../fonts/8x13B.bdf -b 30 -C 255,0,0 -B 255,255,255 --led-gpio-mapping=adafruit-hat

I see white background, in the upper half I see red clock and in the lower half I see blue the same clock.

For clock without background (-B), the clock line single and for upper half - does not have blue, for lower half does not have red.

Could you help?

anp135, 2020-07-14:
[http://wiki.amperka.ru/_media/продукты:rgb-led-matrix-driver-cap:rgb-led-matrix-driver-cap_schematic.png?w=650&tok=645d9f](url)

hzeller, 2020-07-14:
Do they use the same pin-mapping as this project ? If not, then you'd see some artifacts as missing colors.

Check their schematic and see what GPIO pin they connect to what output. Then add the mapping to here:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/hardware-mapping.c

Or maybe they use one of the existing already like `--led-gpio-mapping=adafruit-hat` or `--led-gpio-mapping=classic` ?

anp135, 2020-07-14:
> Do they use the same pin-mapping as this project ? If not, then you'd see some artifacts as missing colors.
In examples they used adafruit-hat mapping, but I suppose something wrong.

> Check their schematic and see what GPIO pin they connect to what output. Then add the mapping to here:
> https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/hardware-mapping.c
Unfortunately I can't read schematic (link upper) for understood GPIO<->HUB75 connectivity -(.
 
> Or maybe they use one of the existing already like `--led-gpio-mapping=adafruit-hat` or `--led-gpio-mapping=classic` ?
--led-gpio-mapping=classic does not work completely (black matrix). I compiled with hardware adafruit-hat and this mapping a bit works:

./clock -f --led-cols=64 --led-rows=32 ../fonts/7x13.bdf -d "%H:%M:%S"  -b 40 -C 0,0,255 -B 255,0,0
upper half: red background, blue clock
down half: black background and red copy clock

./clock -f --led-cols=64 --led-rows=32 ../fonts/7x13.bdf -d "%H:%M:%S"  -b 40 -C 0,0,255 -B 0,0,255
upper half: completely black
footer half: completely blue

./clock -f --led-cols=64 --led-rows=32 ../fonts/7x13.bdf -d "%H:%M:%S"  -b 40 -C 0,255,0 -B 0,0,255
upper half: background black, green clock
footer half: completely blue

./clock -f --led-cols=64 --led-rows=32 ../fonts/7x13.bdf -d "%H:%M:%S"  -b 40 -C 255,0,0 -B 0,0,255
upper half: background black, red clock
footer half: completely blue

./clock -f --led-cols=64 --led-rows=32 ../fonts/7x13.bdf -d "%H:%M:%S"  -b 40 -C 255,0,0 -B 255,0,0
upper half: completely red
footer half: completely black

./clock -f --led-cols=64 --led-rows=32 ../fonts/7x13.bdf -d "%H:%M:%S"  -b 40 -C 255,0,0 -B 0,255,0
red clock upper half and full matrix green background

./clock -f --led-cols=64 --led-rows=32 ../fonts/7x13.bdf -d "%H:%M:%S"  -b 40 -C 255,0,0 -B 255,255,0
upper half: red clock and yellow background
footer half: green completely

anp135, 2020-07-14:
I found this one: http://wiki.amperka.ru/продукты:rgb-led-matrix-driver-cap
here missed E and LAT and D both on the GPIO20. This is looks like adafruit-hat mapping instead strobe GPIO21. I made custom mapping from adafruit-hat and change .strobe from GPIO21 to GPIO20 - completely does not works, all matrix are black

hzeller, 2020-07-14:
Maybe they have a typo in their documentation ? I suggest to mail the Amperka people directly.

anp135, 2020-07-14:
Yes, looks like typo with strobe and should be GPIO21. Anyway can't access to blue in upper and red in footer halfs of matrix.

So you can't help me as I understood?

hzeller, 2020-07-14:
It looks like a hardware issue with  your adapter board or  your matrix, so you have to follow the GPIO connections and see if they make it out of your board. And that your matrix boards are working. So nothing I can help you from here, this is a physical issue you have to check locally.

Of course, you should check also the troubleshooting section in the README, e.g. double checking that you don't have anything else running that accesses your GPIO pins.

anp135, 2020-07-14:
thanks.

I checked with direct GPIO connection and raspberry perfect works with led panel directly. So this is hat hw problems.
thanks.

alealexa, 2020-08-04:
> thanks.
> 
> I checked with direct GPIO connection and raspberry perfect works with led panel directly. So this is hat hw problems.
> thanks.

Please be aware that amperka uses -led-gpio-mapping=adafruit-hat-pwm mode.
Is also should be powered up directly from power supply to particular jack of hat.
Otherwise it will get some power from hub75 connection which enough to light "on" LED, but it wouldn't work properly.

Alexey

anp135, 2020-08-04:
-led-gpio-mapping=adafruit-hat-pwm|adafruit-hat didn't help.
power supply connected to hat and also each panel connected to power directly.