hzeller/rpi-rgb-led-matrix #1101: Matrix code on a different thread -- "Can't set realtime thread priority 99"

ElBertrando, 2020-07-15:
Since I am both polling for changes of the recorder as well as running a constant loop to output to the recorder, I decided to use a new thread for the Matrix logic, as the loop was never exiting. Am I on the right track by tying a new thread for this ? 
I am running the code with 'sudo', but am getting the following error:

[image]

My code for updating the thread which is being done from polling to set the recorder state is: 

```
#!/usr/bin/env python3

import time
import TheRunningText as RunT
from threading import Thread
import queue

argsList = []
class Watcher:
    def __init__(self, recorderState = "Unknown"):
        self._recorderState = recorderState
        global sThread
        global argsList
        argsList = [20,"Starting...",255,0,0]
        sThread = Thread(target=RunT.RunText().runTextWithSpeed, args=(argsList))
        sThread.start()
    def get_recorderState(self):
        return self._recorderState
    def set_recorderState(self, newState):
        global argsList
        if (self._recorderState != newState and newState == "Recording"): 
            print("Sign should be reading Recording now")
            self._recorderState = newState
            argsList[1] = "RECORDING"
            argsList[3] = 255
            argsList[4] = 255
            #RunT.RunText().runTextWithSpeed(20,"RECORDING",255,0,0)
        elif(self._recorderState != newState and newState == "Paused"):
            print("Sign should be reading Paused now")
            self._recorderState = newState
            argsList[1] = "PAUSED"
            argsList[3] = 255
            argsList[4] = 255
            #RunT.RunText().runTextWithSpeed(10,"PAUSED",255,255,0)
        elif(self._recorderState != newState and newState == "Idle"):         
            self._recorderState = newState
            print("Sign should be reading Idle now")
            argsList[0] = 0
            argsList[1] = "IDLE"
            argsList[3] = 255
            argsList[4] = 255
```

hzeller, 2020-07-16:
If you use threads, you should know everything needed about locking, synchronization, signalling etc.
Before attempting threads in any context, you have to be deeply familiar with these concepts.
I can't help about this in Python (I usually write stuff in C++).

The priority 99 thing is if you're not running as root. I suspect you have the runtime set to drop privileges, so once the matrix is started, you don't have root anymore. But not sure how you got there if you use Python threads.

ElBertrando, 2020-07-16:
Right now I don't see a way to simultaneously poll for recorder changes while running a loop to display the text. I am using `self.lock.acquire()` and `self.lock.release()` for the two loops on different threads, but keep getting the same error. As I said I am using sudo to run the code. Do you know where I might run "sudo setcap 'cap_sys_nice=eip' $THIS_BINARY" ??

ElBertrando, 2020-07-21:
I have decided to forgo threading for now due to these problems -- we can close

ElBertrando, 2020-08-13:
Was able to work threading using lock  acquire and lock release to separate out common variables...

closed 2020-08-13 by ElBertrando