hzeller/rpi-rgb-led-matrix #1102: Raspberry pi Zero with active adapter + ENC28J60

arahasya, 2020-07-16:
I have a project which requires LAN with raspberry pi zero
I am going to integrate ENC28J60  with the active adapter for this purpose .
Will the SPI communication with ethernet cause any problems for the rpi-rgb-led-matrix GPIO operations?
GPIO 8,9,10,11,25 are to be used for this communication chip.

Please advice. Thanks!

daveythacher, 2020-08-09:
I am going to make an attempt at this, since I saw it and no one else has. This library uses pretty much all the pins on the GPIO header for three HUB75 chains. So if you only need one or two you can probably make this work. Outside of that you would have to use USB.

What you need to do is remap the pin here: https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/hardware-mapping.c. The only pin to my knowledge that matters is the OE pin it needs to be connected to a PWM pin. The library supports working without it but it is recommended to use it if possible. (More info here: https://github.com/hzeller/rpi-rgb-led-matrix#improving-flicker)

However if using a single chain you may be able to use the Adafruit-hat-pwm version. Based on the comments this appears to avoid the SPI pins and use a valid PWM pin for OE. If this works you do not need to make source changes. You can select this pin with --led-gpio-mapping=adafruit-hat-pwm as described by the documentation: https://github.com/hzeller/rpi-rgb-led-matrix#changing-parameters-via-command-line-flags.

arahasya, 2020-08-10:
Your answers are unrelated to my question.
I know how to do it. Just want to know the effects of running SPI parallel to this lib. Anyway I tried it myself after receiving the hardware and it works.

closed 2020-08-10 by arahasya

daveythacher, 2020-08-10:
> Your answers are unrelated to my question.
> I know how to do it. Just want to know the effects of running SPI parallel to this lib. Anyway I tried it myself after receiving the hardware and it works.

I did answer the question. You asked if it will cause issues for the GPIO operations. Which it will if using default setup and trying to use the affected chain. The Pi uses some logic reductions in many of its peripherals. There are a number of way GPIO vs special function are prioritized. I do not know the exact method for the Pi. However if you remap them there should not be a conflict for this to matter. That mapper avoids driving the pins and generates a mask.

>Will the SPI communication with ethernet cause any problems for the rpi-rgb-led-matrix GPIO operations?

As for the performance it would depend on other factors. My guess is that the using the ENC28J60 kernel driver and IP stack, which will be around 8-10Mbps fully loaded, will not cause an issue. Overhead for this is not too terrible. The overhead of other software using this library and ethernet will matter more. 

You mentioned some of the configuration information, but did not disclose all of it. There is a lot of configuration possible for two chains. However it looks like you are planning to lower the usage quite a bit. However if that ends up actually happening with this library I do not know.

You are driving two chains. Perhaps the default 30 percent usage is for three chains or one. In any case that should not be an issue, but I am guessing. However I do not think this is a fair wording if this is what you meant.