hzeller/rpi-rgb-led-matrix #1103: FM6127 does not work reliably past 2 panels chained?

marcmerlin, 2020-07-17:
I just received some newer panels which turned out had FM6127 instead of FM6126A. They almost work, except for chaining:
[image]

If I put the panel first in the chain, it's all good, but by the time it's 3rd, I pseudo randomly get some sections that do not get initalized correctly and a color component won't turn on.
If I go back to FM6126A with the same rPi4 and cables, it works perfectly, so I think the tming for FM6127A may be off by just a bit?

arahasya, 2020-07-23:
It is most probably a fault in that particular driver IC of that board
Any boards after that will show problems

daveythacher, 2020-09-10:
The FM6126A and FM6127 are very similar it would appear. The FM6127 has LED open detection logic. It also has 3 configuration registers. Which the code has been updates somehow to configure.

Honestly I have no idea how the FM6126A is even working. The FM6127 may be working under the open detection logic mode. The BCM OE logic may cause a few issues. The min time for pwm-lsb-nanoseconds would need to be 40. However that is the same for FM6126A and the default is 130. However the data latch command to my knowledge is not generated.

Smart Matrix Figured out there was a difference between the ICN2038 and ICN2038S. This may explain the FM6126A. I am guessing the configuration bits are from trial and error or a logic analyzer capture from FPGA. Which would create problems. However it would appear the FM6127 works in single panel mode, which I cannot explain.

These are awesome datasheets. They do not document the block diagrams properly. They do not list the configuration registers. They look like they are copied and pasted without a clue. OE is marked as active low in some sections but active high in others.  LE is called SLAT, which other parts do use.

There is a number of different things that could be going on here. In theory there could a be a few glitches possible with BCM OE logic. Then again that would depend on how you read the block diagram. Which does not mention the notion of command states. Since this logic does not generate a single command for the data latch, despite the data latch command existing, who knows what it does with the serial out pin.

Overall technically speaking these are the PWM/MM like chips was talking about. It also randomly mentions grayscale stuff, if you translate the Chinese, which would suggest PWM. These chips use commands from LAT and CLK. They also support configuration settings. Which the initialization logic talks too, and possibly correctly.

It looks like these chips are hybrids of PWM chips and non-PWM chips. Usually OE drives the PWM counters as clock source. I confused myself pretty good due to the quality of diagrams. The command block is not present, and it also mangles the notion of latch pretty good. OE is likely driving combinational logic. So there is likely only to storage levels despite the diagram mentioning three. The ghost/LOD implementation is odd to me based on block diagram, but I am not super familiar with that anyhow.

The reasons why I say this project does not support PWM and MM drivers is mostly due to OE for BCM. Which they could do with LAT and a timer, but that is a problem for Linux. I know how to make the OE clock work and how to workaround the corresponding state issue. The other issue is the missing command logic and configuration bits, which is also straight forward.

Linux will actually have a better time with the PWM and MM drivers from bit banging. However can suffer issues in context switch and memory/system bus stability. But I refuse to commit to this project. I have difference of opinion on how this should be handled in Linux. I am willing however advise some of the approaches I have considered. I have covered some of this for my personal project.

I would recommend attempting to properly send data via data latch command. Honestly not sure why this was not included in the FM6126A and FM6127 changes. At least support of the commands could have been prototyped in.

@gohlhausen Where did you get those configuration bits?

daveythacher, 2020-09-10:
This is interesting: 
https://github.com/gohlhausen/rpi-rgb-led-matrix/blob/e709dee267f6f98b39cb821132bcacae612db06e/lib/framebuffer.cc#L490-L552

daveythacher, 2020-09-10:
There is a lot of potential issues here: The OE BCM, Bit banging, configuration bit errors, timing violations, cable issues, temperature, humidity, etc. All mostly complete unknowns.

I have seen panels do stuff like this and this: https://user-images.githubusercontent.com/7902641/65637900-4e8ed880-dfe5-11e9-9d6a-446825d88011.jpg In fact there is one that does this near where I live.

I would like to see a datasheet which confirms random configuration bit settings. You can get issues from over current or bad limits.

arahasya, 2020-09-10:
With all your knowledge just wondering why don't you solve any issues and create a PR? 
@daveythacher

daveythacher, 2020-09-10:
What you want for the is a configurable structure that is modular. That way you can apply profiles. When you select FM6127 you also select a mapping of this configuration structure during init. This structure is then read at the correct times. 

For example the number of cycles for CLK while LE is high for data latch. Or the configuration register count, the number of cycles of CLK while LE is higher to clock that in, the value of that configuration register, etc. This would allow you to make a structure array with index or pointer to a structure to pass.

Now in dumpToMatrix in framebuffer.cc you have something like:
for (int col = 0; col < columns_ ++col)
{
     // Get data
     // Set data and clear clock
     if (col >= columns_ - ptr[index].dataLatch)
          io->SetBits(h.strobe);
     // Set clock
}

if (ptr[index].dataLatch)
    io->SetBits(h.strobe);
io->ClearBits(h.strobe);

Ever get Deja vu? For me its nearly always like someone is trying to ruin my life.

marcmerlin, 2020-09-10:
As an update to this original thread, I wasn't able to reproduce this reliably. I think there was something electrically weird that caused issues. For now, I have an array where I replaced an FM6126A with an FM6127, and things work fine.
Whatever problem I saw was likely a borderline effect that may not happen if I swap panels around or use a different cable.
On the plus side (for me), my current application will be 3 panels connected in parallel, so I don't have to worry as much about chaining.
Otherwise, I'll second @arahasya : @daveythacher  get yourself some hardware if you haven't already, try out everything you've been saying, and submit a PR if you have improvements. PRs >>> ideas in bugs :)

marcmerlin, 2020-09-16:
[image]
[image]

marcmerlin, 2020-09-16:
what's interesting between the 2 pictures is that it shows the init string that works reliably on FM6126A, is off my enough to fail to init some portions of the screens, not always the same, and sometimes the last panel gets portions initialized even though they're not correctly initialized on a panel before it.

daveythacher, 2020-09-16:
@marcmerlin I think these issues have been common on both.

Lets try something. Use max pwm-lsb (3000) and gpio-slowdown (5). Move pwm-bits to 2.

marcmerlin, 2020-09-17:
@daveythacher not sure if that should have helped, but no difference outside of worse refresh rate

root@rPi4:~# ~/rpi-rgb-led-matrix/examples-api-use/demo --led-gpio-mapping=regular --led-rows=64 --led-cols=128  --led-row-addr-type=0 --led-chain=4 --led-show-refresh --led-slowdown-gpio=5   --led-pwm-bits=2 --led-panel-type=FM6126A  --led-parallel=3 --led-pwm-dither-bits=2 --led-pwm-lsb-nanoseconds=100 -D4 --led-pixel-mapper=V-mapper

I did 3000 nanoseconds too, but the refresh rate is so terrible that I can't use it at all.

marcmerlin, 2020-09-17:
[image]
[image]
[image]

daveythacher, 2020-09-17:
https://github.com/hzeller/rpi-gpio-dma-demo/blob/master/img/rpi2-direct-loop.png

daveythacher, 2020-09-17:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/97ec43d07f3f85431db4d4a29836fd6a412b0c68/lib/gpio.cc#L338
I would try this:
```C
  slowdown_ = slowdown * slowdown * slowdown;
```
I would also use -D5 with gpio-slowdown 5. We no longer need to mod pwm-bits or pwm-lsb-ns.

marcmerlin, 2020-09-17:
Mmmh, this really looks like shooting in the dark, you're cubing the value, that's a huge slowdown.
How are you doing on the front of getting yourself some panels? 
Which connection board are you using? passive? active? passive-3? active-3?
What hardware have you setup for yourself and gotten working so far?
Do you want a source of FM6127 panels so that you can get your own and try all the code you'd like against them?

marcmerlin, 2020-09-17:
https://www.amazon.com/dp/B0869P1DCH is now using FM6127

daveythacher, 2020-09-18:
> Mmmh, this really looks like shooting in the dark, you're cubing the value, that's a huge slowdown.
> How are you doing on the front of getting yourself some panels?
> Which connection board are you [...]

Let me try this one more time:
There is a couple types of time issues. The time and non-time timing issue. I am going to assume this is a non-time timing issue, otherwise we would be screwed as the chip is defective. Both however should respond favorably to slowing it way down.

The longer version of this goes back to electrical theory, frequency, signals, bandwidth, EMI, transmission lines, TTL, capacitors, gate delays, temperature, bit banging, etc. If you look at that picture it shows basically a sine wave not a square wave. This is hell for TTL. Granted those likely have Schmitt trigger inputs.

Basically low pass filters can form on wires, pins, etc. which block fast signals. It can cause random behavior in the narrow TTL region. The Pi is generating a clock at an unknown frequency. This frequency could be stable or unstable. The LED drivers have a max recommended of 25MHz. Some drivers will do more 30+MHz for example, while others will be closer to the recommended value. Normally I see a lot of the configurations from the FPGAs using 18MHz.

Long wires need stronger drivers. Which is why the active boards exist. However these drivers are included in the input of every panel. These boost across panel and to next panel. If the layout or shielding is poor cross talk can cause random data on the wires. Also impedance can start to mess around.

As I understand it the first board in the chain works. However the later boards not so much. This to me suggests a signal issue. Possibly too much interference, too fast of clock, and/or too little drive strength. Finding this is easy using a scope. However I am not sure you have one or that we need one just yet. I am also assuming you would have used it if you had it.

Therefore we do the fall back. Just slow it down. Now this is a consequence of bit banging and not having the clock set by a hardware peripheral which will not vary as much. There is a lot of ways to use existing logic, assuming better logic existed; and adding logic to make this work. That never happened for this project.

hzeller's GPIO testing does appear to suggest coming out of the Pi, 25MHz should be good enough. Therefore I have to assume the clock could be higher. Random behavior down the line suggests problems, remember some drivers work better than others. However all will work at recommended conditions. I guess it could be power but that makes little sense to me on one hand. I would check board order, cables, power, speed, etc. However I am going to assume that has been done already.

Therefore we slow down to see if there is a failure point with the clock. Simple enough?

I have no interest in acquiring FM612x boards. I have no need for them. I could run additional tests if I had them, but at this time there is little reason for me to do so. I also dislike how this project manages the configuration bits. Just randomly setting bits that I cannot verify with a datasheet or some other verifiable documentation, bugs me. I think I know of a way to work around into getting it, but I make no promises at this time.

There is a bug that can happen with the power based on how it does OE BCM. (Different from the bandwidth issue.) When you put a large load against a cap the voltage sags, which can cause issues too. These are the main power rails for the LEDs and LED drivers, which can cause hysteria conditions. hzeller recommends using lots of caps randomly to prevent this but he can cause high inrush current. There is a way to handle this, but it sounds like I should keep that to myself.

I attempted to bring these issues up several times. Not completely sure I am right, but this is what I would look into.

marcmerlin, 2020-09-18:
@daveythacher  I hear you, this is not my project, but having done open source for around 30 years, this is not how things work.
You can't just contribute code and ideas you've never tested.  Talking theory is all nice and good, but at the end of the day, it's just words and you need to test it if you want people to pay attention.
If you want to contribute, you really need to get yourself some hardware, test what you're saying, and then submit a patch, or in this case you could send me a patch to test to check that what seems to have fixed the issue for you, also works for me (because not all panels are identical).
So, please test, and if you aren't interested in doing so/don't want to get the hardware, lots of theory in bugs or ranting that you'd have done it a different way, isn't really that helpful, sorry.

daveythacher, 2020-09-18:
Lets be clear here: This is bug classification.

I have sent you a prospective test for clarification. Honestly I would have expected you to have done this already.

arahasya, 2020-09-18:
I hate to say this but this has become quite irritating now. @daveythacher You aren't the owner of this library. But you have hijacked almost all the issues with your long posts with 0 outcome. 

Maybe have those long discussion in the PR sections and only refer them in the comments in the issue section.

daveythacher, 2020-09-18:
@arahasya I would understand that if it were not for the fact that almost every post of yours is basically pointless. You do not even "rant". There is no information that is conductive. There are solutions within the "long discussions" at least. 

How long has the FM612x driver issues existed?  You have maybe two people working on it, and neither seem to have a clue. So when someone does come along you tell them to spend their time and money to help you do nothing. Then you come up with this wacky narrative, to justify such a thing. I am not going to spend my money fixing your problem. You should be doing that. Look I have tried to be nice, but honestly there are about 3 or 4 people that have no business being on github.

Many of the issues here are related. They are all solvable! However you will end up rewriting the library or spending money on hardware. Which will likely be rejected. Part of the issue is I have seen projects like this before. This project conceptually has a weird interface and fails to disclose important information correctly.

The statements made about DMA alone are enough to keep certain people away. I regard much of this as misinformation and not within ethical guidelines. I have also found the responses of this project to be outside expectation. There are complete jerks that at least serve some valid purpose in their posts, but I cannot say that about some users here.

I have read forms on other places of the internet were people have thought more about this library as complete novice users of it. People of engineering like background will quickly spot the issue with the BCM period. That defect is giving you hell. The exact manner is not known, but should be really easy with a scope or logic analyzer to track down. 

Could it be fixed perhaps. However the issue is it will still be subject to a problem. This project is based around shift register LED drivers, which should be done with short chains. This project tried to do things for cheap, but makes many reckless claims. So I create "fix" which will not solve the issue and will likely be misunderstood. Heck what I have said is said to be incorrect, somehow.

I did not understand how you were doing BCM. In reality you do a BCM with stretch period, which mostly destroys your lower bits. You have some percentage of the pwm bits worth of color steps. Don't worry I am not going back to that. However there is a power issue possibly because of that. However my statements about this were in reference to "naive" PWM. 

The concurrency of shifting in parallel against a shared power rail is questionable practice.  This is further compounded with incorrect notions of BCM period. I would expect if this was the case the first bits in a shift register would be corrupted, while the last would be correct. I would expect the first bits to be at the end of the chain.....now does that sound familiar? Nope because you have probably not read this!

Look in another post I was having a long discussion about this implementation against certain claims. However the assertion here is you really smart and I do not have clue. Believe me the only one who should be irritated is me. 

You can get a lot of this issue controlled with MM LED drivers, which I have attempted to point you toward repeatedly. However those are not supported. They are harder to find, debug, etc. However they are more conductive to bit banging from user space on shared cache. Or you can emulate the function that you want, but then you have to do "expensive" and "complex" hardware. 

Never mind you could make it all software and update it from the Pi. Just make dumb hardware, that gets loaded by software.

Instead we have semi-spaghetti C++ OO code. I think the layout is chaotic. I think the library integrates features in the wrong locations. I think C++ is a little forced in certain places. It reminds me of Java more than C++.

So in summary: slow it down or stabilize your rails. Either way I could not give darn, what you think or do.

daveythacher, 2020-09-18:
Both of these have ghosting logic. Which is a capacitance thing. I would study the configuration bits.

arahasya, 2020-09-18:
Instead we have semi-spaghetti C++ OO code - Did you change one line of this code to improve the repo?

I think the library integrates features in the wrong locations. I think C++ is a little forced in certain places. - Did you improve any of this code? 

And if you feel like it's "wasting your time" to send a PR then you don't know how GitHub works.

daveythacher, 2020-09-18:
> Instead we have semi-spaghetti C++ OO code - Did you change one line of this code to improve the repo?
>
> I think the library integrates features in the wrong locations. I think C++ is a little forced [...]

You mean a rewrite? That is not a pull request or bug fix.

arahasya, 2020-09-22:
well, Henner has an open source library which thousands of people have successfully used. All your ideas may be good but aren't worth anything if you don't devote your "time" without expecting anything in return like Henner has done. Or just create your own library which is superior. So if you aren't ready to do anything on your own and just expect Henner to improve the library with your "superior knowledge" might as well pay him to do that.

daveythacher, 2020-09-22:
I am responding to this against my own advice. Maybe I need to use bullets?

- OE is implemented in PWM to avoid real time problems.
  - Still has problems with real time, since next BCM bit can be slow.
    - Results in extended period which distorts the color representation.
  - Assuming it was implemented/configured correctly.
    - Implementation suggest it was really meant for shorter chains originally.
  - Really used to save cycles and keep track of time
    - Really all hardware would prevent issue completely.
      - MM panels are expected to be more successful for this reason
    - Mappers sort of complicate this
      - Hardware would fix this, even a wire jump would.
- OE in PWM creates power issues. 
- Power issues create problem for FM612x panels in long chains.
  - Large power loads.
- "Solution" is to use lots of caps.
  - Large caps are slow but more powerful.
    - You can stage them, but that may not be needed too much here.
    - Panels may stage some for you.
  - Has issue with frequency created by BCM or switch current for logic and drivers.
    - Multiple switch edges per period for BCM
      - Cap lifespan could be shorter with this.
    - Single switch edge per period for PWM or "naive" PWM as this project refers to it sometimes.
- Large inrush due to large caps from power supply.

Issue here comes from large chains. Pi is not conductive to that with shift registers via bit bang. Hence the implementation of BCM with PWM. Which exaggerates the power issues. Thus the question is can it be fixed...or should it be fixed?

If you want to use a Pi....you can. You just need to be smarter about it. Use the right panel type. Use a decent library/approach. Use the right hardware. That is what I was trying to explain to you!

The MM version of this library will likely want to be a complete overhaul to actually track the panel architecture internally. The OO structure would follow that for support sake. If you even decided to bother with OO to begin with.

For your review: https://docs.github.com/en/github/site-policy/github-community-forum-code-of-conduct

alsaleem00, 2020-09-26:
I have similar issue here with FM6127.
It looks the initialization sequence is not propagated well to the second unit.
The first one in-line is working ok, but not the next (it is dimmed). I reversed the order and still the same.

daveythacher, 2020-09-26:
@alsaleem00 Can you mention your power supply and chain length. As much setup information as possible. I am only replying since you appear new here. Lot of trolls here, fyi.

Just to use #1160 as example:
There are people on here driving possibly 8 to 24 64x128 RGB panels using a 64A power supply. Then throwing capacitors on the board. It will work for the average current. However you will not be able to use all 64A since those caps will cause some overhead.

In that setup you could potentially have 276A in Peak current. The caps will attempt to normalize this to lower than 64A average current. However you have a frequency problem, which is effectively duty cycle of a sort. Assuming 10mOhm wire you will experience huge wire loses. Adding a huge cap will create a low pass filter which will block fast frequencies. I seriously doubt the math for capacitor sizing is not taking most of this into account.

There are solutions. Use more power supplies. Use power supply with higher power rating to get it close to the peak power. Shorter or thicker cables. Using high voltage to low voltage switch mode regulators. Stage high cap banks on input high side with inrush.

Or do something really crazy like lower the current gain inside the FM612x panels. See https://github.com/pixelmatix/SmartMatrix/issues/78

These have active logic inside which will cause issues if it browns out. Normally this does not matter. It could behave in a number of ways.

Now if you know what you were doing or are using smaller chains then most of this works just fine. However I believe there is a defect inside the logic. I think BCM exaggerates problems like this. Again when used correctly the problem may not matter. This code base likely targets smaller panel setups. It targets shift registers, which have limits they themselves impose. There are newer panels which work around this explicitly.

alsaleem00, 2020-09-26:
I am using 2 of 126x64 (P2.5). The power supply (for both) is 10A. 
The panels are mounted on a lexan plate and chaining them by a 30cm ribbon.
The distance from controller to matrix is about 40cm. I am using breadboard with 74AHCT245 to level shift signals.
The problem has been solved. I just re-adjust wires going to matrix.

Thank you for the links. They are very helpful.

daveythacher, 2020-10-05:
@marcmerlin You use a 60A 5V supply right? Not the recommended 122.88A 5V power supply?

arahasya, 2020-10-21:
Facing similar issue. Need help from you Henner, @hzeller 

Data is propagating to exactly 7 FM6127 individual IC's properly, After that it is either blank or garbage.

I have manufactured the panels myself so have done all possible tests and swapped the cabled and panels. 

Definitely issue with FM6127.

arahasya, 2020-10-21:
worked when I used slowdown-gpio=5

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828