hzeller/rpi-rgb-led-matrix #1104: Any advice on how to remove this flicker?

jerkwagon, 2020-07-18:
Hi everyone, I have read over the troubleshooting parts and still can not seem to fix this flicker. 

Here is the P5 32x64 panel i am using [Link to panel](https://www.aliexpress.com/item/32841046668.html?spm=a2g0s.9042311.0.0.5fdc4c4dNWDqwl)

[image]

I am using a Pi3 with raspian lattest version and updates.  I think i will try to install the lite version, as this noise isnt going to suffice for my project.   I also have a nano laying around i could try as well

Looking for any input, thanks in advance

side note , is there anywhere i can get general community support for LED panels , discord or forum?  not much on reddit..

hzeller, 2020-07-18:
You didn't describe how you connected the panels, did you use the standard wiring, or some Adafruit HAT ? If you used the hat, did you do the PWM modification ? If not, you absolutely want to do that.

The glitches seen there are mostl likely due to
   * Not using PWM (hence the first question)
   * Pi to fast for panels. Adjust the slowdown gpio value.
   * Power supply has trouble. Add capacitors and make sure ot have short wiring.

hzeller, 2020-07-18:
As for forum, I think @marcmerlin was interested in setting up something like that, but I don't have time for that. For now, the bugtacker is a perfectly good place.

jerkwagon, 2020-07-18:
> You didn't describe how you connected the panels, did you use the standard wiring, or some Adafruit HAT ? If you used the hat, did you do the PWM modification ? If not, you absolutely want to do [...]

sorry about that, here is a photo of my wiring, 
[image]

no adafruit hat,

GPI slowdown didnt really change much

i noticed gpio slowdown is best a 0 or 2,  on the volume mixer demo i dont get that much noise, its very noticeable on the video player.  

my goal is to run 3x2 matrice of 32x64 matrix boards to make  video frame of sorts. 

one thing i have not done is ground the Ground pins, since my

jerkwagon, 2020-07-18:
turns out i was missing my ground. issue fixed. 

@hzeller  thanks for your time i really appriciate it!

closed 2020-07-18 by jerkwagon

marcmerlin, 2020-07-18:
@hzeller  I just asked Louis if he can help with his existing web board.

hzeller, 2020-07-19:
Glad you got it working @jerkwagon .

In particular I like the progress of debuging, with one comment ending in a dangling
```
one thing i have not done is ground the Ground pins, since my...
```

... and the next continues with
```
Turns out i was missing my ground. issue fixed.
```

:)