hzeller/rpi-rgb-led-matrix #1105: rpi-rgb-led-matrix with rpi computer module

rajahslimen, 2020-07-20:
hey there 
i'm using rpi computer module
does any one test this library with rpi cm3
or it will work ?

arahasya, 2020-07-23:
Yes
Can get 6 parallel chains using compute module
Will soon put up the hardware on sale that I have developed for the Compute module

hzeller, 2020-08-17:
Now, support is there, provided by @arahasya in PR #1055 

You have to enable the wide GPIO support [in the lib/Makefile](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/Makefile#L138)
by removing the comment on the wide GPIO comment

```Makefile
DEFINES+=-DENABLE_WIDE_GPIO_COMPUTE_MODULE
```

There is no breakout board yet, you''ve to wire it up manually for now.

hzeller, 2020-08-17:
You find the mapping on the [alternative hardware mappings](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#alternative-hardware-mappings) paragraph: there is a fold-out table at the bottom ("GPIO-pins for each hardware mapping").

Or here:

|         | regular | adafruit-hat | adafruit-hat-pwm | regular-pi1 | classic | classic-pi1 | compute-module |
----------|---------|--------------|------------------|-------------|---------|-------------|----------------|
Parallel chains|        3|             1|                 1|            1|        3|            1|               6|
~OE       |GPIO 18  |GPIO 4        |GPIO 18           |GPIO 18      |GPIO 27  |GPIO 0       |GPIO 18         |
Clock     |GPIO 17  |GPIO 17       |GPIO 17           |GPIO 17      |GPIO 11  |GPIO 1       |GPIO 16         |
Strobe    |GPIO 4   |GPIO 21       |GPIO 21           |GPIO 4       |GPIO 4   |GPIO 4       |GPIO 17         |
A         |GPIO 22  |GPIO 22       |GPIO 22           |GPIO 22      |GPIO 7   |GPIO 7       |GPIO 2          |
B         |GPIO 23  |GPIO 26       |GPIO 26           |GPIO 23      |GPIO 8   |GPIO 8       |GPIO 3          |
C         |GPIO 24  |GPIO 27       |GPIO 27           |GPIO 24      |GPIO 9   |GPIO 9       |GPIO 4          |
D         |GPIO 25  |GPIO 20       |GPIO 20           |GPIO 25      |GPIO 10  |GPIO 10      |GPIO 5          |
E         |GPIO 15  |GPIO 24       |GPIO 24           |GPIO 15      |        -|            -|GPIO 6          |
Chain 1/R1|GPIO 11  |GPIO 5        |GPIO 5            |GPIO 11      |GPIO 17  |GPIO 17      |GPIO 7          |
Chain 1/G1|GPIO 27  |GPIO 13       |GPIO 13           |GPIO 21      |GPIO 18  |GPIO 18      |GPIO 8          |
Chain 1/B1|GPIO 7   |GPIO 6        |GPIO 6            |GPIO 7       |GPIO 22  |GPIO 22      |GPIO 9          |
Chain 1/R2|GPIO 8   |GPIO 12       |GPIO 12           |GPIO 8       |GPIO 23  |GPIO 23      |GPIO 10         |
Chain 1/G2|GPIO 9   |GPIO 16       |GPIO 16           |GPIO 9       |GPIO 24  |GPIO 24      |GPIO 11         |
Chain 1/B2|GPIO 10  |GPIO 23       |GPIO 23           |GPIO 10      |GPIO 25  |GPIO 25      |GPIO 12         |
Chain 2/R1|GPIO 12  |             -|                 -|            -|GPIO 12  |            -|GPIO 13         |
Chain 2/G1|GPIO 5   |             -|                 -|            -|GPIO 5   |            -|GPIO 14         |
Chain 2/B1|GPIO 6   |             -|                 -|            -|GPIO 6   |            -|GPIO 15         |
Chain 2/R2|GPIO 19  |             -|                 -|            -|GPIO 19  |            -|GPIO 19         |
Chain 2/G2|GPIO 13  |             -|                 -|            -|GPIO 13  |            -|GPIO 20         |
Chain 2/B2|GPIO 20  |             -|                 -|            -|GPIO 20  |            -|GPIO 21         |
Chain 3/R1|GPIO 14  |             -|                 -|            -|GPIO 14  |            -|GPIO 22         |
Chain 3/G1|GPIO 2   |             -|                 -|            -|GPIO 2   |            -|GPIO 23         |
Chain 3/B1|GPIO 3   |             -|                 -|            -|GPIO 3   |            -|GPIO 24         |
Chain 3/R2|GPIO 26  |             -|                 -|            -|GPIO 15  |            -|GPIO 25         |
Chain 3/G2|GPIO 16  |             -|                 -|            -|GPIO 26  |            -|GPIO 26         |
Chain 3/B2|GPIO 21  |             -|                 -|            -|GPIO 21  |            -|GPIO 27         |
Chain 4/R1|        -|             -|                 -|            -|        -|            -|GPIO 28         |
Chain 4/G1|        -|             -|                 -|            -|        -|            -|GPIO 29         |
Chain 4/B1|        -|             -|                 -|            -|        -|            -|GPIO 30         |
Chain 4/R2|        -|             -|                 -|            -|        -|            -|GPIO 31         |
Chain 4/G2|        -|             -|                 -|            -|        -|            -|GPIO 32         |
Chain 4/B2|        -|             -|                 -|            -|        -|            -|GPIO 33         |
Chain 5/R1|        -|             -|                 -|            -|        -|            -|GPIO 34         |
Chain 5/G1|        -|             -|                 -|            -|        -|            -|GPIO 35         |
Chain 5/B1|        -|             -|                 -|            -|        -|            -|GPIO 36         |
Chain 5/R2|        -|             -|                 -|            -|        -|            -|GPIO 37         |
Chain 5/G2|        -|             -|                 -|            -|        -|            -|GPIO 38         |
Chain 5/B2|        -|             -|                 -|            -|        -|            -|GPIO 39         |
Chain 6/R1|        -|             -|                 -|            -|        -|            -|GPIO 40         |
Chain 6/G1|        -|             -|                 -|            -|        -|            -|GPIO 41         |
Chain 6/B1|        -|             -|                 -|            -|        -|            -|GPIO 42         |
Chain 6/R2|        -|             -|                 -|            -|        -|            -|GPIO 43         |
Chain 6/G2|        -|             -|                 -|            -|        -|            -|GPIO 44         |
Chain 6/B2|        -|             -|                 -|            -|        -|            -|GPIO 45         |

arahasya, 2020-09-08:
Finally received the active-6 PCB. Will populate soon. Need to slightly change the pin out from what is given above. Hopefully everything works out :) @hzeller 

[image]

KentWalker, 2020-09-09:
Hi @hzeller,
Is there a reason for keeping GPIO0 and GPIO1 out of the hardware mapping with the CM3+ module?  I would like to free up GPIO14 and GPIO15 as TXD0 and RXD0 if possible

arahasya, 2020-09-09:
Well it is technically possible to use GPIO 0 and GPIO 1
But I have used GPIO 31 for LAN_RESET which 

But if you have a bulk requirement I can make that custom hardware for you.

rajahslimen, 2020-10-17:
Hi @hzeller,
I did try to use your lib with rpi cm3 with 3 parallel chains and it work perfectly 
I wanna ask what do i have to add or change to have 6 parallel chains?

hzeller, 2020-10-17:
First, you have to enable the wide GPIO by enabling `ENABLE_WIDE_GPIO_COMPUTE_MODULE` in the Makefile. Remove the [comment character in front of line 138 in `lib/Makefile`](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/Makefile#L138). And recompile.

Then, you can use the `--led-gpio-mapping=compute-module` flag to enable the compute module mappin.. In the fold-out table in [the alternative hardware mappings](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#alternative-hardware-mappings) section, you see the mapping of GPIO pins to LED panels (also see above in one of the comments: there is the table as well).

Then you can use `--led-parallel=6` and use all chains. I have not been able to test that yet, but @arahasya has made a board and tested it I believe.

arahasya, 2020-10-17:
Hi Henner, 
I have realised I made a mistake in the active-6 board. I used the 3rd buffer line of OE CLK and LAT for the 3 extra chains.
Should have buffered separately for each chain. Will rectify it.

andytheengineerguy, 2023-07-19:
Hi @arahasya, I made a [post in the discourse forum](https://rpi-rgb-led-matrix.discourse.group/t/curious-has-anyone-implemented-a-cm4-solution/702) about this topic, but I'd yet to come across this thread, where I've noticed you've developed an active 6 board already :) 

Do you have a link to this board? Are you selling it? Curious to hear back from you as I am trying to develop (or potentially buy) the same thing 👌

Ry-Kode, 2024-01-16:
Hi @arahasya,
I have the same question : are you selling this board? Or willing to share the design? I'm too trying to connect 6 chains

Jan1503, 2025-01-18:
Hey guys,
I'm thinking about the CM4 and if I could use it with 6 chains to build another big matrix.
After further reading I realized that the CM4 does not expose the needed pins so I think I have to use the CM3?!

There is a breakout board from waveshare, will this one work?
[https://www.waveshare.com/wiki/Compute_Module_IO_Board_Plus]https://www.waveshare.com/wiki/Compute_Module_IO_Board_Plus)

Thanks for comments!

marcmerlin, 2025-01-19:
Well, I was talking to Chow Me from electrodragon about this ( @info@electrodragon.com ) and he was saying that for CM4, the pins exposed are already used for other things like wifi, RAM and other such needs.
Apparently it's possible for CM3 but it seems that the CM3 modules may be out of production, or at least hard to find/get, so he was not feeling too hot about making a board for a CM that may not be possible to buy anymore.
Have you found that CM3 is still reasonably available and will continue to be?

marcmerlin, 2025-01-19:
@arahasya  did you ever got your board to work? Are you able to make a few more and sell them?  :)

Jan1503, 2025-01-19:
I just bought two CM3+ 16GB for 19€ each here in Germany and will order the Waveshare board later today.

As you said, the CM4 is not usable because the gpios are used for ETH and this stuff. Maybe you could use the version without wifi and solder directly on the pads on the pcb but thats far from ideal.

I need ETH for my matrix to feed the videostreams so do you know if a usb-to-ethernet adapter might work?
Too bad, the waveshare board doesn't have a dedicated ethernet onboard.

marcmerlin, 2025-03-26:
@Jan1503 sorry for the slow answer: yes USB ethernet adapters should work as long as they are supported by linux.
Were you able to make progress?

closed 2025-11-27 by marcmerlin