hzeller/rpi-rgb-led-matrix #1106: Accessing background color in python bindings

ElBertrando, 2020-07-21:
I have a pretty good working model now. I am trying to access the background color in the DrawText function (from the runtext.py) python example. I know you added `canvas->Fill(bg_color.r, bg_color.g, bg_color.b); ` for me when I was running by script, but I don't see a way access it via samplebase.py or from `DrawText()`

hzeller, 2020-07-22:
In the Python binding, there is also a `Fill(red, green, blue)` method on the canvas that you can use.

ElBertrando, 2020-07-22:
Using **runtext.py** as basis, I should be able to access the `Fill()` method when the `offscreen_canvas = self.matrix.CreateFrameCanvas()` is created. So I should be able to access the method by `offscreen_canvas.Fill(R,G,B)` ? However I do not see autofill in my IDE so I must be doing something wrong in this situation...

hzeller, 2020-07-22:
Well, apart from the IDE, did you run the code and see if Python complains ?

hzeller, 2020-07-22:
Read the C++ code how that outline frame is done: you first need to [generate an outline font](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/utils/text-scroller.cc#L169) that is essentially a larger font, one pixel around the original font. Then [draw that first in your desired outline color](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/utils/text-scroller.cc#L215) before you draw the regular text.

ElBertrando, 2020-07-27:
I am looking at the graphics.pyx code and see that it takes 6 arguments for the python bindings:
```
def DrawText(core.Canvas c, Font f, int x, int y, Color color, text):
    return cppinc.DrawText(c.__getCanvas(), f.__font, x, y, color.__color, text.encode('utf-8')
```
However the c++ code takes 8 arguments to modify the DrawText method. Do I need to modify this ?

hzeller, 2020-07-28:
you might need to add the kerning parameter to the Python bindings.

Turmburger, 2020-07-28:
Bbnnkojjj
Nudeln uikn, z zu if
⁣Gesendet mit BlueMail ​

ElBertrando, 2020-08-13:
I have a fully working project now :) However I am just now revisiting the outline problem as my last task. Just to clarify, I will need to modify the graphics.pyx code to add the kerning parameter -- Just this file ? 
Also I cannot read the German in Turmburger commented

closed 2025-11-27 by marcmerlin