hzeller/rpi-rgb-led-matrix #1107: What GPIO pin mapping should be used to connect a 32(rows)X64(cols) RGD LED Matrix with raspberry pi4

mushfiq-heraclius-xen, 2020-07-24:
I have recently connected my 32(rows) X 64(cols) RGB LED Matrix with Raspberry Pi4  by following the Table "Table: GPIO-pins for each hardware mapping" according to the "regular" connection scheme as below:
R1----GPIO_11
G1----GPIO_27
B1----GPIO_7
R2----GPIO_8
G2----GPIO_9
B2----GPIO_10
OE----GPIO_18
LAT/STB---GPIO_4
CLK----GPIO_17
A----GPIO_22
B----GPIO_23
C----GPIO_24
D----GPIO_25
is the pin mapping i followed is correct or do i have to change it to other scheme to use raspberry pi4 ??
i can compile the demo codes and dont get any compilation error.
i tried "./demo -t 10 -D 1 runtext.ppm" but nothing shows up on display, still its dark.[i also provided --led-rows=32,--led-cols=64  too and i tried --led-slowdown-gpio=4] but nothing happens.
then whenever i try to run the "minimal-example.cc" under "example-api-use" directory i get the compilation error, error is "no such header led-matrix.h exists"
 please help me and guide me regarding what pin mapping scheme should i use and how to remove that compilation error that is led-matrix.h not found error!  i want to run the  "minimal-example.cc" , Thank in Advance @hzeller

hzeller, 2020-07-24:
Yes, Raspberry Pi 4 should work unless you have a really really old version of rpi-rgb-led matrix.
You want to run the demo with sudo (`sudo ./demo ...`)

If you get an issue with the header, something is messed up; how did you get the source-tree ? I suggest to do a fresh checkout and recompile.

```
cd ~
git clone https://github.com/hzeller/rpi-rgb-led-matrix.git matrix-new
cd matrix-new
make
cd examples-api-use
sudo ./demo ...
```

mushfiq-heraclius-xen, 2020-08-05:
@hzeller at last i found the issue with my hardware gpio mapping setup, the jumper cable i used to pass CLK signal was broken so i replaced with a new jumper cable ,hence i was able to light up the display and run all of your codes successfully, my mistake was i was trying to run the "makefile" command inside the "examples-api-use" directory which was a completely wrong approach. Then i realized that i have to run the "make"command only once in root directory and i get the whole library working, but the only issue i was having is the upper 16 row of my 2 panels , the upper 16 row's LEDs are glowing blue all the time, as according to the  above mentioned gpio mapping some of the pins are shared with the SPI bus too. I have disabled SPI of my rasp-pi4 on config.txt too, but the blue color issue of upper 16 rows still remained. its always stay high as the B1 data line of my RGB LED Matrix is connected with gpio7(CE1)  , how can i set the GPIO7(CE1) pin to normal gpio mode?? TIA.

hzeller, 2020-08-05:
It _should_ just behave as a normal GPIO by default. If not, did you double check that you don't have a damaged Pi ?
I once destroyed a couple of GPIO pins by accidentally mis-plugigng an extension board...

mushfiq-heraclius-xen, 2020-08-05:
i tried the "adafruit-hat" gpio pin mapping scheme and that worked perfectly but , i'm unlucky that i cant make it work with "regular" gpio-pin-mapping scheme, all of gpios are oki, i tested with individual LED blink test too!
and here is the config.txt of my rpi-4, please help :) 
[

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# w1-gpio 1wire interface
#dtoverlay=w1-gpio

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=off
]

mushfiq-heraclius-xen, 2020-08-06:
finally solved the problem with regular gpio mapping, it was my bad luck that the GPIO expansion breakout board i used has a missing/break in the connection from the 40 pin male box socket to a breakout pin , and that pin was GPIO7(CE1)  that drives the data line B1.
[image]
[image]
finally had to solder a flying jumper wire to fix the connection , hence now i'm able to run my RGB LED Matrix smoothly with no glitch or flicker.  Thank you @hzeller as u told me to double check all the connections thus i found the problem. :) 
i have also use 74LS245 as level translator just like u did in the adapter board too.
[image]
[image]
and i really admire your contribution for the library, the library is indeed a very good one to follow. I have learnt so many things. thank you once again  💯

hzeller, 2020-08-06:
Yay, glad you got it working!
Have fun with your LED matrix!

closed 2020-08-06 by hzeller