hzeller/rpi-rgb-led-matrix #1109: Modification of makefile of video-viewer for WiringPi

Turmburger, 2020-08-01:
Hi,

I would like to use the GPIO, so I use WiringPi. According to the error report I have to modify the makefile:

make -C ../lib
make[1]: Verzeichnis „/home/pi/rpi-rgb-led-matrix/lib“ wird betreten
make[1]: Verzeichnis „/home/pi/rpi-rgb-led-matrix/lib“ wird verlassen
g++ -O3 -W -Wall -Wextra -Wno-unused-parameter -D_FILE_OFFSET_BITS=64 video-viewer.o -o video-viewer -L../lib -lrgbmatrix -lrt -lm -lpthread -I/usr/include/arm-linux-gnueabihf -lavcodec -lavformat -lswscale -lavutil
/usr/bin/ld: video-viewer.o: in function `LIES_PIN()':
video-viewer.cc:(.text+0x3a0): undefined reference to `wiringPiSetup'
/usr/bin/ld: video-viewer.cc:(.text+0x3ac): undefined reference to `pinMode'
/usr/bin/ld: video-viewer.cc:(.text+0x3b8): undefined reference to `pinMode'
/usr/bin/ld: video-viewer.cc:(.text+0x3c0): undefined reference to `digitalRead'
/usr/bin/ld: video-viewer.cc:(.text+0x3d0): undefined reference to `digitalRead'
/usr/bin/ld: video-viewer.cc:(.text+0x3e0): undefined reference to `digitalWrite'
/usr/bin/ld: video-viewer.cc:(.text+0x3e8): undefined reference to `delay'
collect2: error: ld returned 1 exit status
make: *** [Makefile:35: video-viewer] Fehler 1

Thanks for support
Joachim

hzeller, 2020-08-01:
You are not linking the wiring pi library, this is why you get linking errors.

But, trying to use another library that also accesses the GPIO will probably not work well, so it will likely not work (or only either the LED matrix or the thing you want to do with wiringPi will work).

What do you want to do ?

hzeller, 2020-08-01:
If you need inputs, then you should use the features provided by rpi-rgb-led-matrix itself: https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/examples-api-use/input-example.cc

Turmburger, 2020-08-02:
Thanks for the link! I want to modify the speed of a video by a sensor signal.

Turmburger, 2020-10-14:
Thanks. Further questions came up -> other thread

closed 2020-10-14 by Turmburger