hzeller/rpi-rgb-led-matrix #1110: Keep getting address issues on my 64x32 led matrix

skggs, 2020-08-01:
Total beginner, trying to build this scoreboard ( https://github.com/riffnshred/nhl-led-scoreboard) and can't seem to get the connection to the matrix right, keep getting address issues. Its an Adafruit led 64x32 screen with a HUB75 port (not hub75e). Ive checked the hardware a dozen times and can't seem to find any issues there. Any help would be appreciated a ton!

hzeller, 2020-08-01:
What do you mean with "address issues" ?
How do you connect the LED matrix (manually, using Adafruit Hat, own adapter ?)

skggs, 2020-08-02:
I just get random patterns on my screen when I try to run the sample. I’m using an adafruit hat

hzeller, 2020-08-02:
did you choose the hardware mapping for the adafruit hat ?

skggs, 2020-08-02:
yeah, so this the exact command I'm running cd nhl-led-scoreboard/submodules/matrix/bindings/python/samples
sudo python3 runtext.py --led-rows=32 --led-cols=64 --led-gpio-mapping=adafruit-hat --led-brightness=60

hzeller, 2020-08-02:
I'd try first the non-Pyhton test programs to reduce the things that can go wrong. Once you get the [demo program running](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#running-some-demos), you can move on to testing Python if needed then the scoreboard.

Also make sure to adapt the gpio slowdown settings (might be needed if you run a fast Pi such as the Pi4) and read the [troubleshooting section](https://github.com/hzeller/rpi-rgb-led-matrix#troubleshooting).

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828