hzeller/rpi-rgb-led-matrix #1112: Image Viewer Shrinking Square Images

camow7, 2020-08-06:
When I use the image viewer to display images, non-square signs (e.g. 2x8 panels) display perfectly.However on square configs (e.g. 3x3 panels or 96x96 pixels) the image gets shrunk to a 32x32 pixel image? It won't fill the 96x96 pixel area?

hzeller, 2020-08-06:
What are the parameters to the image viewer (all the `--led-*` options) ? How big is your image (e.g. is it >= 96x96 pixel so that it can be scaled to the size?)

camow7, 2020-08-06:
This is the command I'm using:

sudo ./led-image-viewer --led-rows=32 --led-cols=32 --led-chain=9 --led-parallel=1 --led-slowdown-gpio=4 --led-pwm-bits=6 --led-show-refresh --led-limit-refresh=130 --led-brightness=80 animation.gif

Have tried with a 96x96px gif and a 500x500px gif. Only tested gifs at the stage. Works fine if we take the same pis and plug it into a 32x704px sign.

hzeller, 2020-08-06:
The parameter you gave just define a large 288x32 pixel screen, so the image is scaled to fit within that, where it hits the 32 pixel high limitation; so a square image will be 32x32 pixels.

If you have them arranged in some sort of 3x3 grid to get a 96x96 physical layout, you need to tell the library by passing in the remapping information, have a look at
https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#remapping-coordinates

(BTW, for such large panel I strongly suggest to use parallel chains to get more refresh speed.)

camow7, 2020-08-06:
You're right. I noobed. 

For your information we ran a 22 panel sign on a raspberry pi 4 and got a clean 130hz with 6bit on a single chain. It runs really well. 90% on core 3. We are switching to 3 chain so it runs at full colour but thought I would note that on an RPi 4 the performance is impressive!

closed 2020-08-06 by camow7