hzeller/rpi-rgb-led-matrix #1113: Unable to  find the hat

arash1221, 2020-08-12:
Hello my feirend,
I couldn't able to find the hat which you mention from electrodragon website, Can I replace it with these ones?
1-adafruit raspberry pi bonnet
2-adafruit  raspberry pi  hat
are these hat works for chain the panels?

[image]
[image]
[image]

depili, 2020-08-12:
https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/

Did you look at all to the raspberry pi hat section of their product list?

-- 
This message may contain traces of Truth.

arash1221, 2020-08-12:
yes,my problem is that I don't have visa,or mastercard to buy,this link you provide only accept these two payment method only. which I don't have!!

hzeller, 2020-08-12:
The hat will work, just make sure to set the correct [--led-gpio-mapping command line option](https://github.com/hzeller/rpi-rgb-led-matrix#if-you-have-an-adafruit-hat-or-bonnet)

arash1221, 2020-08-13:
Thank you @hzeller,  which board? the first black one(adafruit bonnet)
or the second one the blue one?
and also  can I chain the leds with this module?

daveythacher, 2020-08-15:
Both of these are handled by adafruit-hat. So you would add --led-gpio-mapping=adafruit-hat to the command line. Note these define all 5 address lines in the pin mapper meaning those pins should not be used without changing the pin descriptions.

Talks about Adafruit boards:
https://github.com/hzeller/rpi-rgb-led-matrix#switch-the-pinout

5 address line stuff:
https://github.com/hzeller/rpi-rgb-led-matrix#64x64-with-e-line-on-adafruit-hatbonnet

The HAT:
https://www.adafruit.com/product/2345
https://cdn-learn.adafruit.com/assets/assets/000/022/441/original/leds_schem.png?1421437847

The bonnet: 
https://www.adafruit.com/product/3211
https://cdn-learn.adafruit.com/assets/assets/000/051/031/original/adafruit_products_schem.png?1518648935

The pin descriptions:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/13bb51bd95ef843f2fb48b4e333f39c6b4a70ebe/lib/hardware-mapping.c#L73

Recommendation for closing.

daveythacher, 2020-08-15:
> and also can I chain the leds with this module?

This explains how to do this. Some wording is a little weird.
https://github.com/hzeller/rpi-rgb-led-matrix#changing-parameters-via-command-line-flags

With these connectors you will only use chain. Others support up to three. 
Set the --led-rows to rows of each panel.
Set the --led-columns to columns of each panel.
Set the --led-chain to number of panels linked.
Set the --led-slowdown-gpio if using faster pi. (Good luck on this.)
Set the --led-panel-type=FM6126A if panel uses it.
Set the --led-multiplexing if it does not work. (Good luck on this.)
Set the --led-row-addr-type if it does not work. (Good luck on this.)
Set the --led-pwm-bits to improve refresh rate. (Good luck on this, and it is probably lying to you.)
Set --led-show-refresh to see what it is printing in terms of FPS, however this does not mean the color depth is true.
Set --led-pwm-dither-bits to improve FPS if needed. (Good luck on this.)
Set --led-pwm-lsb-nanoseconds to something around 50-300. (Good luck on this. This does not work correctly based on my understanding.)

For performance tweaks against FPS:
https://github.com/hzeller/rpi-rgb-led-matrix#performance-improvements-and-limits

arash1221, 2020-08-15:
thanks alot  @daveythacher,  very  useful  guide

closed 2020-08-15 by arash1221