hzeller/rpi-rgb-led-matrix #1114: 32x64 panel has strangely distorted output.

russweiher, 2020-08-12:
Not sure what's going on with my panels.  "./demo -D1 runtext.ppm" scrolls smooth, and seems to have the correct colors, but the images are strangely interlaced but with the lines in the wrong order or something?  

Here's what I've tried:

- re-install of DietPi
- all of the gpio slowdown options
- all of the multiplexing options
- all of the mapping options
- 2 different control boards (adafruit hat and the 3 channel board from electrodragon)
- 3 different LED panels from the shipment I got
- 2 different power supplies (5v 4a wall adapter and 5.4v 40a power supply)

I've attached 2 pictures of the output I'm seeing.

Thanks in advance for any help you can provide,
Russ

[image]
[image]

hzeller, 2020-08-12:
This might have a different chipset than the more common panels; you have to track down the chip set, read the datasheet and provide a patch to the library to support.

russweiher, 2020-08-12:
Thanks for the reply.  Of the chips on the back of the panels, most of them say "TC5020A", which comes up in google as a 16-bit constant current LED driver.  Are those the chips I need to find the specs for, or would it be one of the other (smaller number) of chips?

hzeller, 2020-08-12:
This is certainly one of the relevant chips. But the 'double line' stuff indicates also some different wiring going on in there; maybe something happens on both clock up and down transition. I'd ask the manufacturer of the matrix itself if they can provide a datasheet how to control the resulting panel.

daveythacher, 2020-08-12:
Here is the datasheet: https://www.alldatasheet.com/datasheet-pdf/pdf/1146426/FUMAN/TC5020A.html

It seems to me you could make a simple program which drive a single line. Then toggles the rows to see how it moves. You could also send different patterns. Play around with different OE drives. You get the point. You would want this to be slow so you can see. Videos of it could be helpful. Take some of the guessing out.

Try runtext16.ppm and/or telling it the number of rows you have. By default it may use three chains, while you appear to only be using one. You can specify that too. What it does with it exactly I am not sure. Overall I think you should check your configuration/command line arguments.

russweiher, 2020-08-13:
Thanks David.  I had found that spec sheet and the one for the FM6126A (and converted from Chinese to English) to see what differences I could find.  While I am a computer programmer, I am not an electrical engineer, and don't have a lot of experience with this low level hardware stuff.  All of the LED matrix panels I've purchased in the past "just worked", so I guess I've just been lucky in that regard.

I like your idea of doing a super basic program to try and test the panel, though I'm actually pretty surprised that doesn't already exist somewhere already.  Surely other people have had to brute force their way into finding the correct settings in the past as new panels and chipsets came out.

daveythacher, 2020-08-13:
I do not read Chinese, but many are written basically the same. Also the information in the drawings and tables is mostly what I am after anyhow. This is a common architecture, but there are a few curve ball versions out there.

I looked at the FM6126A datasheet:
https://www.alldatasheet.com/datasheet-pdf/pdf/1145498/FUMAN/FM6126A.html
Which according to this is wrong:
https://bobdavis321.blogspot.com/2019/02/p3-64x32-hub75e-led-matrix-panels-with.html
However he is showing a LE waveform I did not find in the version I found of FM6126A.

I have seen that command table thing before. That is how a lot of the memory mapped drivers work. Which is not supported by this project. So there is a couple integrated with some limited support. They ignore most of the function control registers. You usually see the command block annotated on the block diagram. However sometimes they forget to put the command table in the datasheet.

Again I would look into your row setup. Once that is ironed out there is a little experimentation with some of the settings.

Here is a command to try: ~/rpi-rgb-led-matrix/examples-api-use/demo 
--led-rows=32 --led-cols=64 --led-chain=1 --led-parallel=1 --led-show-refresh --led-slowdown-gpio=2   --led-pwm-bits=3 --led-panel-type=FM6126A  --led-pwm-lsb-nanoseconds=500000 -D4

Page 14 is interesting for the TC5020A.

russweiher, 2020-08-13:
David - 

I don't read chinese either, but if you look at the HTML version of the spec, and if you are using chrome, you can right-click on the screen and it will give you the option to translate to English.

I tried the command you suggested, though the max lsb-nanoseconds it will allow is 3000, which certainly strobes pretty good.  I took some video of it though, and it almost seems like the screen is divided into 4 segments.  The 1st and 3rd segment light up, and then the 2nd and 4th segment.  Some of the frames show as solid, and some of the frames show a line skipped. When I set it to 100 I get a pretty smooth transition of colors (at 3 pwm bits), and the whole screen is filled.

I was able to find some really basic bit-banging python code here: https://learn.adafruit.com/connecting-a-16x32-rgb-led-matrix-panel-to-a-raspberry-pi/experimental-python-code

I adapted it to the newer PI pinout and added a D control line.  I can get it to pretty much draw the columns as intended (the horizontal spacing of the 3 rectangles matches the given coordinates), but the rows are goofy.  Depending on where I tell it to start a rectangle on the Y axis I might get a solid shape or I might get skipped lines.

Also: On the link you provided, page 14 for the TC5020A datasheet seems to just be the physical specs of the package?

russweiher, 2020-08-13:
Looks like I might have found the solution.  I used "led-row-addr-type = 4", and it seems to give the correct output.  It was a total accident, as the readme page says that value only goes upto 3, but I was just incrementing numbers and testing.

Thanks for all the help!

daveythacher, 2020-08-13:
I meant to say page 4, sorry about that. I forgot LED multiplexing may not show up well in cameras. 

Glad to hear it is working.

daveythacher, 2020-08-15:
This can be closed?

commit 08167221 (2020-08-15): Update ranges of flags in documentation.

Fixes #1114
closed 2020-08-15 by hzeller