hzeller/rpi-rgb-led-matrix #1115: Adapter Board Active 3, OE pin is not connected with any GPIO

mushfiq-heraclius-xen, 2020-08-13:
@hzeller hello sir, i have a question, in the schematic of "adapter board active  3" i have seen no gpio is used for driving OE signal, instead it was kept high by connecting 3.3V through a 10k resistor then passing this signal to all 74HCT245 chips. But according to wiring table it's mentioned that for regular hardware pin mapping i have to tie OE signal pin of RGB LED Matrix Display with GPIO18 of rpi.  whats the magic here ?? can you explain ??

hzeller, 2020-08-13:
`OE` is connected to Pin 12 of the header, which is GPIO 18.
It is also pulled high to make sure the signal is high when the output is not driven, this is the 10k pull-up you see.

The signal is then buffered via the HCT245 and send to the LED matrix headers.

mushfiq-heraclius-xen, 2020-08-13:
@hzeller  thank you very much i just figured it out a while ago :)

closed 2020-08-13 by hzeller