hzeller/rpi-rgb-led-matrix #1116: 74LS245 or 74CT425 or 74HCT245 whats the major difference ??

mushfiq-heraclius-xen, 2020-08-13:
@hzeller i have used 74LS245 ICs , is it a big mistake ?? is there any major difference between 74LS245 and 74CT245 or 74HCT245 ??

hzeller, 2020-08-13:
The reason we use the 245s is to convert the 3.3V logic level to 5V logic level acceptable by cmos inputs.

So the important thing is that the logic family can accept 3.3V logic level as input and output 5V compatible output (which means at least 3.7V to be acceptable by cmos inputs on the matrix panels).
LS does *not* qualify as its output not guaranteed to be at least 3.7V.

Check the datasheet of your logic families or find graphical representations like http://www.interfacebus.com/voltage_threshold.html to make a decision.

closed 2020-08-13 by hzeller

andygrace, 2020-08-13:
They’re pin compatible but critical are the logic levels.  CMOS can run at a range of voltages but LS TTL are designed for 5 volts in and out.  
Look at the data sheet of the particular chip for its characteristics.

LS    = Low power (compared with the really old 74XX series) with Schottky transistors at outputs and other critical junctions which effectively cleans up the signal.
HC   = High speed CMOS 
HCT = High speed CMOS compatible with TTL input/output logic levels.

LS series chips are all bipolar TTL internally - they’re older technology and use a bit more power ie Emitter-Base-Collector transistors 
CMOS uses field-effect transistors and uses less power     ie Source-Gate-Drain transistors.

Henner used these very basic 8 bit register chips to buffer the signal from 3V3 to give 5V and a clean fast output to the LED boards.

mushfiq-heraclius-xen, 2020-08-13:
@andygrace  thank you very much very informative

explit7, 2021-11-09:
I had here a very strange problem, which was solved with replacing 74HCT245 with 74LS245

https://github.com/skiselev/8088_bios/issues/12