hzeller/rpi-rgb-led-matrix #1118: Improve frame rate limit logic

daveythacher, 2020-08-13:
I was looking through the code to see how stable performance of a single core would be. I came across this and it looks like it is not very helpful.
https://github.com/hzeller/rpi-rgb-led-matrix/blob/c8f5a83be52d400ab21c223e829a11754a3dd2a7/lib/led-matrix.cc#L134

My guess is this is done to prevent flicker or due to stability issues caused be system context switching. However I also found this, which suggests there is no way to apply a CPU priority to a single core system.

https://github.com/hzeller/rpi-rgb-led-matrix/blob/c8f5a83be52d400ab21c223e829a11754a3dd2a7/lib/led-matrix.cc#L407

In RTOS there is a few different approaches for normalizing tasks. 
- Making critical processes that yield high priority. 
- Making free running non-critical tasks low priority.
- Making processes yield and setting a priority accordingly
- Etc.

Here it would appear to me you could make this a high priority task if you carve up the load and yield. Fixing the limit logic to yield would obviously create yielding. However this only works if it has enough time to move away to do something meaningful.

The issue here is how to do this. In reality you would probably need to blank the display while switched away. Then putting the thread too sleep. Linux should switch away rather than preempting randomly assuming it obeys priority like a RTOS. (It could do this without being a RTOS.)

You would want to reduce the time spent drawing a frame, so lowering the pwm-bits, using small rows/columns, and increasing the GPIO speed. This will case the frame rate to spike so then you would apply a frame rate limit. These limited frames are where the other processes run.

Granted this is a bit much to explain in the documentation but at least it could be made available if it works.

daveythacher, 2020-11-12:
OE controlled by PWM solves some of this. Assuming it is used.

daveythacher, 2020-11-12:
Looking through this code shows it is possible to make the thread high priority:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/c8f5a83be52d400ab21c223e829a11754a3dd2a7/lib/thread.cc#L48-L75

So now you are on single core, with high priority blocking thread.

daveythacher, 2020-11-12:
Block there and in BCM loops. For BCM in the beginning will be insignificant amount of time but for MSBs it will be larger. How much larger depends on configuration, column length, etc. IO updates are critical, waiting for time to pass is not.

Maybe this is the best policy for gain the kernel to work with you, but I doubt it. Then again that assumes the kernel works like that.

newcokesucks, 2022-09-29:
I am not completely sure this will stop ghosting. As I understand it the gates on the row drivers may not have time to discharge. The frequency between rows could still be too much. The overall refresh is limited but not the multiplexing.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin