hzeller/rpi-rgb-led-matrix #1120: Improve/Explain OE logic

daveythacher, 2020-08-13:
Currently you have OE logic limited to range 50-3000ns. Why? This is not likely required, the current logic supports 16 to 16384. You could improve this to support just about anything in 8ns steps.

It looks like you use the 500MHz clock to drive the PWM, but scale it down to 125MHz. (I am guessing this is need?) The OE value is divided by four which produces the divider for the PWM logic. You then set the PWM value and period as two raised to the BCM bit multiplied by two. So you send a 100 percent duty cycle non repeating pulse to OE. You use inverted logic to handle active low.

You could use 500MHz clock divided by four. Then you could use period and value set to the OE value divided by eight. 
The alternative is the divider based on lsb-ns value. You could do simple division tests to find a scaler.

While your at it you could optimize a few things.
https://github.com/hzeller/rpi-rgb-led-matrix/blob/c8f5a83be52d400ab21c223e829a11754a3dd2a7/lib/gpio.cc#L625

This really needs a comment: (Divide by 8? Why so many?)
https://github.com/hzeller/rpi-rgb-led-matrix/blob/c8f5a83be52d400ab21c223e829a11754a3dd2a7/lib/gpio.cc#L641-L648

hzeller, 2020-08-13:
The comment is there, describing that we [want to minimize each of the entries](https://github.com/hzeller/rpi-rgb-led-matrix/blob/c8f5a83be52d400ab21c223e829a11754a3dd2a7/lib/gpio.cc#L635-L636) as [we have to accomodate empty slots](https://github.com/hzeller/rpi-rgb-led-matrix/blob/c8f5a83be52d400ab21c223e829a11754a3dd2a7/lib/gpio.cc#L651-L666)

The code is based on observing the actual behavior of the limited documentation available and measuring. If you see improvements and can measure the results, send a pull request.

daveythacher, 2020-08-13:
What I am curious about is why that is needed? You only enter that section is working on bit plane four or below. For these you effectively write a period and value of zero or one. You do this 8 times. If I am not mistaken pwm_range_ will only have 2, 4, 8, 16, 32, 64, 128, 256, 512, 1024, 2048.

Why do you send in 8 values for a 16 deep FIFO? Why do you send in zero or one when the hardware cannot handle values < 2?

I am not sure if it will like zero values.  You could expand the period and set it to low value, it would likely be insignificant. You may also try using clear and/or disable logic.

daveythacher, 2020-09-10:
You could use SPI for this also. That would allow default audio to continue to work. It would completely waste it and several pins.

daveythacher, 2020-10-01:
Note --led-pwm-lsb-nanoseconds should not exist the way it does. For most of these panels the serial data clock is recommended to be no more than 25MHz. This would make the OE time 40ns at its most fundamental for a single bit serial register. 

This setting should be automatically computed to be 40 times the number of columns in nanoseconds. I see no reason why this value is configured by the user which will likely be confusing and/or wrong. Note this does not fix the drift issue that can appear.

The issue here is the FPS would be reduced potentially. However I would argue quality reduction has already happened via the color representation.

There is still drift in the bit banging which is the actual problem. I guess this argument here is for accurate representation of quality against FPS. Which would allow the user to make changes to PWM (BCM) bit depth to improve FPS while getting the most quality they can. Granted this current approach my be trying to turn into the wave to avoid being tipped, however this again is potentially unneeded.

That being said I think the library for small column number and a properly aware user can use the --led-pwm-lsb-nanoseconds setting to fine tune the behavior. However proper documentation does not exist. In fact even asking about this was meet with scorn.

daveythacher, 2020-10-08:
Here is a small program which computes the max PWM bits and recommended OE time in nanoseconds for a given FPS, display size and serial clock. This could be done automatically at boot using defaults or hard coded values for FPS and serial clock.

For example:
Given: 64x8 columns, 32/2 rows, 95 FPS, 25MHz serial clock, 11 Max BCM/PWM bits
Result: 5 bits, 640ns OE
Recommended command:
--led-cols=64, --led-rows=32, --led-chain=8, --led-pwm-bits=5, --led-limit-refresh-95, --led-pwm-lsb-nanoseconds=640

I would consider the led-limit-refresh optional. Note this is based around a notion of serial clock which is not verified or known. It can change based on CPU load. The processor could overspeed or under speed. Logic could detect this based on frame time.

I have a suspicion that this code base is intended for shorter chains. However serial clock put hard limit on max length and PWM bit values regardless. Not even a FPGA could fix that. Overall this project attempts to use high clock to average away instability, which will work. However it works best for shorter or smaller problems. Just like the power line instability.

```C++
#include <iostream>
using namespace std;

int pow(int x) {
        int result = 1;
        for (int i = 1; i <= x; ++i)
                result *= 2;
        return result;
}

int log(int x) {
        int result = 0;
        while (x) {
                x = x / 2;
                ++result;
        }
        return result;
}

void compute_bits(int columns, int freq, int max_bits, int rows, int fps, int *bits, int *oe_ns) {
        int val = freq / (columns * fps * rows);
        double oe = (double) columns / (double) freq;
        oe /= (double) pow(max_bits - log(val - 1) - 1);
        oe *= 1000.0;
        oe *= 1000.0;
        oe *= 1000.0;
        *bits = log(val - 1);
        *oe_ns = (int) oe;
}

int main(int argc, char **argv) {
        int bits, ns;

        compute_bits(8 * 64, 25 * 1000 * 1000, 11, 16, 95, &bits, &ns);

        cout << "Bits " << bits << " OE NS: " << ns << endl;

        return 0;
}
```

This is related to #918 and #971

daveythacher, 2020-10-08:
FPGA drivers have advantage of high pins. Some can drive around 16 chains, while this project uses nearly all pins to drive 3. The performance of a chain is governed by serial clock. High density panels require doing the following:
- High serial clock
- Huge parallel bus
- Optimized PWM locality
- Optimized image data locality

All of these reduce the data bandwidth. Data compression would be the next step. However compression algorithms could make hardware expensive and could have undefined worse case.

A code base could exist for using new drivers which exploit the locality points listed above. However this would likely become confusing in terms of interface and code structure.

daveythacher, 2020-10-08:
You can get the serial clock by measuring the time this code takes:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/0dfce9121572c998b7a85d4836c13b9de05e61f9/lib/framebuffer.cc#L856-L860

daveythacher, 2020-10-08:
So what you can do is measure the serial clock. That will allow you to either derive the FPS according to PWM bits or derive the PWM bits according to FPS. 

Overall I think the OE notion is confusing and pointless for most settings. Perhaps you disable all of this is that is set. This would be reserved for people that knows how it works. There can be a few conditions you could want this. However for most people it is a problem.

Currently you indirectly derive FPS based on PWM bits and OE. This usually is wrong, but can be configured correctly if understood. However little documentation exists.

daveythacher, 2020-10-08:
Overall the point the attached outline is to, by force, configure the library correctly. Since the library uses logic that is a little tricky and not well documented.

There is two modes outlined in it:
- You give FPS and scale PWM to meet it.
- You give PWM and FPS scale to meet it.

A third mode is possible if you default OE ns to illegal value and disable this logic when set. (Not shown)

Part of the reason this logic is not documented and this is left as settings for user is due to GPIO bit bang. Performance here is tied directly to serial clock frequency which can vary. Therefore the logic outline here attempts to regulate/detect this frequency and scale either FPS or PWM to meet expected targets.

Another topic would be discussing power limits by either using current limiters, correct power supplies or limiting the number of on LEDs. These two issues appear a lot in the tracker beside unsupported hardware configurations.

All of which are solved by documentation!

This is really rough:
[#1120.txt](https://github.com/hzeller/rpi-rgb-led-matrix/files/5345661/1120.txt)

daveythacher, 2020-10-15:
You can do crazy stuff with this by the way. Like attempt to move period to less than power of two steps. This is useful for regulating bandwidth. However the disadvantage of serial panels without PWM or MM is this directly reduces quality. However for smaller or cheaper setups that likely to use this, this may not be a problem.

To active larger setups with better quality you will need to move to hardware solutions. FPGA, MM LED driver, etc.

Despite this the documentation for how to configure or use this logic is not present. I have seen much confusion about this.

daveythacher, 2020-11-13:
MarcMerlin added this gem:
https://github.com/hzeller/rpi-rgb-led-matrix#performance-improvements-and-limits

The LSB time is OE_time * (2 ^ (11 - pwm_bits)). So if you select 7 bits of PWM with 100ns. The time for the LSB will be 1.6uS. Now if shifting 128 columns using 25MHz this will take 5.12uS. Which means will loose the lower 2 to 3 bits of the 7 bit PWM. To fix this you increase the OE_time to 320. However that will reduce your FPS proportionally. By roughly a factor of 3.2 times.

Want to get a boost in performance? Use 5 bits PWM with 100ns. This will allow 192x128 on 3 chains. Which is about 24K of pixels or 8K per chains.
Want more pixels? Use 4 bits PWM with 100ns. This will allow 192x256 on 3 chains. Which is about 48K of pixels or 16K per chain.
Want more pixels? Use 3 bits PWM with 100ns. This will allow 192x512 on 3 chains. Which is about 96K of pixels or 32K per chain.
Want more pixels? Use 2 bits PWM with 100ns. This will allow 192x1024 on 3 chains. Which is about 192K of pixels or 64K per chain.
Want more pixels? Use 1 bits PWM with 100ns. This will allow 192x2048 on 3 chains. Which is about 384K of pixels or 128K per chain.

Will the Pi hold...probably not. Will the multiplex ratio time against FPS break down...probably. Will the color depths be decent...no.

The interesting thing is these LED drivers like parallel connections. FPGAs can do that, but the Pi cannot. The interesting things is there are optimizations available to the Pi which can starve off the need for a FPGA. However the Pi will struggle to take full advantage of that. All in all the Pi is only so interesting. I mean FPGAs make huge LED walls all the time and they get so much more FPS and color depth.

Apples to oranges I would say to that. Why anyone would bother with that I am not sure. The Pi has its merits so long as you keep it sane.

daveythacher, 2020-11-13:
The Pi is not likely pushing 25MHz by the way. It depends on a number of factors what it actually does. Pi model for example has huge effect on it. Surrounding work loads. So the idea he is getting 7 bits of PWM on 16K pixels per chain is absolutely shocking.

By the way FPGAs are not interesting since they cannot generate as he says more pixels than a Pi....but a FPGA can generate 25MHz. Memory is the issue with FPGA. Clock is the issue with Pi. Context my friend context. Location location location.

daveythacher, 2020-11-13:
The assumption is that you need PWM bits, which is interesting. There is a possible debate over how the code base should handle that in terms of mappers. However it would be pointless. 

Setpixel for example has overhead for PWM bits which if not used is wasting cycles. So in some sense it should be a different code base or use different implementation all together.

Granted the amount of time require to process 128K pixels per frame is high. Pushing you into multithreading which could case performance issues. More pixels causes issues regardless for Pi. So does generating extra PWM bit steps you will waste.