hzeller/rpi-rgb-led-matrix #1121: Problems with 1/8 Scan and out of order rows

James2k2, 2020-08-15:
I've been trying to get my 32x32 panels to work and no matter what I do, I can't get the row orders to align correctly.

Setting multiplexing option 1 gives me the best image but as can be seen from the photo below the middle 2 sets of 8 rows should be swapped with the top and bottom 8 rows.
[image]

hzeller, 2020-08-15:
mmh, so if there is none of the 16 multiplexing options work, we'd need to add a new one, possibly a kind of StripMultiplexMapping but with flipped second row. Let me have a look.

James2k2, 2020-08-15:
16 multiplexing options? Apologies I can only access 10 of them.

If it helps I compiled using the Adafruit guide, I'm assuming it's out of date? I'll also have a look around. Thank you!

hzeller, 2020-08-15:
Please use this particular library here directly https://github.com/hzeller/rpi-rgb-led-matrix and follow the instructions in the README. The Adafruit people are using my library, but are always a little bit behind in syncing them.

Make sure to set the `--led-gpio-mapping=adafruit-hat`

James2k2, 2020-08-15:
Recompiled and I've tried all 16 now, no dice unfortunately. Option 1 still gives me the best result so far. If it helps, I've attached a few photos of the panels I've got. Thank you again! 
[image]
[image]
[image]

hzeller, 2020-08-15:
Can you do a fresh `git pull` and recompile, testing if the option 17 I just added works ?
It flips the way the parts are addressed:
[image]

new one
[image]

hzeller, 2020-08-15:
Oh, and just realized there was a typo. Now it is what it should be
[image]

(count was off by one)

James2k2, 2020-08-15:
Well sir, you have done a stirling job! Thank you very much. Here's a screenshot. 
[image]

hzeller, 2020-08-15:
Yay, have fun with your LED panels!

closed 2020-08-15 by hzeller

James2k2, 2020-08-15:
OK, you're going to hate me now! The order seems to be swapped again when I add a second panel. These panels are very confusing. Also, please do just tell me if these panels are too problematic to fix! 
[image]

hzeller, 2020-08-15:
What are the parameters you are passing ? Did you use `--led-chain=2` ?
It should be something like
  `--led-cols=32 --led-rows=32 --led-chain=2 --led-multiplexing=17`

James2k2, 2020-08-15:
You know what, it's late and I've completely missed that parameter! I do apologise. Yes it works perfectly! Now to try and get Adafruit Framebuffer code to accept the extra mode you added. Thank you again for your patience and doing my stupidity checks!

hzeller, 2020-08-15:
I think the rpi-fb-matrix just uses this project as sub-module.
So you probably just have to

```
cd rpi-rgb-led-matrix
git checkout master
git pull
```

hzeller, 2020-08-15:
mmh, they use a deprecated transformer that doesn't exist anymore, so that code won't compile.

So it is probably easier to just hack this in: In your rpi-fb-matrix/ directory, edit `rpi-rgb-led-matrix/lib/multiplex-transformers.cc` and modify line 74; replace the `<`
```
    const bool is_top_stripe = (within_panel_y % (panel_rows_/2)) < panel_rows_/4;
```
with `>=`
```
    const bool is_top_stripe = (within_panel_y % (panel_rows_/2)) >= panel_rows_/4;
```

.. then use `--led-multiplexing=1`

James2k2, 2020-08-15:
Well you're a fountain of knowledge. Thank you very much! I had attempted to look through your examples to see if I could modify their code to use a newer transformer but it wasn't presenting itself to me at this moment.

hzeller, 2020-08-15:
Essentially what needs to be done is to implement a [`PixelMapper`](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/pixel-mapper.h) instead of a tranformer. It should be somewhat straight-forward; both map pixels from source to destination essentially.

The PixelMapper requires a little less brain-twisting (but it means that the old transformer has to be un-brain-twisted).