hzeller/rpi-rgb-led-matrix #1122: Green LEDs not working in upper half (32x64)

devdass, 2020-08-17:
I decided to pull my panels and RPi 3B out of storage and see if I could still get the demos to run.

First I wired it up and ran a demo. The upper half of the panel wasn't displaying any green. I re-wired and encountered the same issue. I connected another panel up in series, and that panel wasn't displaying green either. 

I then tried a different RPi (3B+) and encountered the issue again!

I have 2 more panels to try but re-wiring is very tedious. 

Any advice?

devdass, 2020-08-17:
I recompiled the git clone directory, and now the green works! haha! However the left half of my panel seems to have intermittent power issues.

Edit: I just tweaked my gpio-slowdown and it seems perfect now.

:D

hzeller, 2020-08-17:
If you wire up the other panel you have as first panel, how does that work ?
It sounds like it is not the Pi, as you replaced that. Also you checked the wiring multiple times.
If you connected the second panel after the first panel, maybe it is just the green wiring on that panel that doesn't work (and then also not propagates?)

Also, it sounds like you're wiring manually: be aware that many panels need 5V logic level and will only work non-reliably with the 3.3V logic level coming from the Pi.

hzeller, 2020-08-17:
Cool! Have fun with your LED panels.

closed 2020-08-17 by hzeller