hzeller/rpi-rgb-led-matrix #1123: Help me ,I have 2 panels of 32*64

6010110254, 2020-08-18:
and i am using adafruit-hat with the Pi4 when running  on rpi-rgb-led-matrix/examples-api-use  
sudo ./demo -D 4 runtext.ppm 
output design here . red 2 rows [scroll]
[image]
[image]
[image]

where sholud i fix it?

hzeller, 2020-08-18:
Did you systematically go through the options such as `--led-row-addr-type`	and`--led-multiplexing`  ? Since you are using the Pi4, you probably also need the `--led-slowdown-gpio` option.

Please read and follow the []README](https://github.com/hzeller/rpi-rgb-led-matrix) first and follow all instructions there systematically.

6010110254, 2020-08-18:
I've tried it
sudo ./demo --led-cols=32 --led-rows=32 --led-chain=2 --led-row-addr-type=2 --led-multiplexing=1 --led-slowdown-gpio=4 --led-pixel-mapper=U-mapper -D 1 runtext.ppm

But results like this
[image]
[image]
[image]

Help me thank you very much

hzeller, 2020-08-18:
Why are you using u-mapper, when you only have one panel provided ?
The panel is 64 columsn wide, so `--led-cols=64` is probably more appropriate.
Did you go through all four row-addr-types, two led-panel types and 17 led-multiplexing ?

If so, your panel probably is not yet supported and you need to hunt down a datasheet for it.

6010110254, 2020-08-18:
I have tried to fix it. --led-row-addr-type 2 The only option to see results and --led-multiplexing has tried all 10 values and it seems to see the results. But still not clear
[image]
[image]
[image]

Please take a look at this LED p5.
https://www.igetled.com/product/94/p5-fullcolor-%E0%B9%83%E0%B8%99%E0%B8%A3%E0%B9%88%E0%B8%A1

Thank you very much, sorry for wasting your time.

hzeller, 2020-08-18:
Do you have a picture of the input connector on the matrix ?
I suspect that this is an entirely different panel (or you plugged in the input on the output :) ).

hzeller, 2020-08-18:
If you only see 10 values for led multiplexing, you're using an ancient version of this library. did you check out the Adafruit fork ?
I'd use this one here

daveythacher, 2020-08-19:
--led-gpio-mapping=adafruit-hat

6010110254, 2020-08-19:
I have removed and reloaded the version, but still the same, tried multi 0-17 and I fixed the adafruit-hat.
But I power the 5v LED through the adafruit-hat board, don't know if that's right.
[image]
[image]
[image]
[image]
[image]

6010110254, 2020-08-19:
[image]
[image]

6010110254, 2020-08-19:
After I connected to a switching 5V, the results were quite dramatic, but still not clear.
[image]
[image]
[image]
[image]

6010110254, 2020-08-19:
I feel the led rows in the middle are not showing.
Why

6010110254, 2020-08-19:
[image]
 thanks very much

daveythacher, 2020-08-19:
This can be closed

closed 2020-08-19 by 6010110254