hzeller/rpi-rgb-led-matrix #1124: Issue with 16x32 panel

manuel1507, 2020-08-19:
Hi, first of all congratulations for this excellent work! I'm trying to work with the library and a bunch of 16x32 LED panels purchased on Aliexpress (P10). My HW setup: Raspberry PI 3 B + Adafruit HAT and DietPi (OS) and one panel for now. I made some steps forward, but I got stucked here
[image]

played around for a few hours with all different settings, but only minor progress. The picture shows my best effort with the runtext python sample :-) 
Any idea what I'm doing wrong here?

thanks a lot!

daveythacher, 2020-08-19:
Good info, but I would need more information like how many panels and your command. So lets start here:

~/rpi-rgb-led-matrix/examples-api-use/demo --led-rows=16 --led-cols=32 --led-chain=1 --led-parallel=1 --led-show-refresh --led-slowdown-gpio=2 --led-pwm-bits=3 --led-pwm-lsb-nanoseconds=300 -D4 --led-gpio-mapping=adafruit-hat

For this first pass I am ignoring a few things: Multiplex, mapping, addr_type, driver special case, etc.

manuel1507, 2020-08-19:
Thank you David for the support. I followed your suggestion and I can
appreciate some improvements on the performance, but still rows from 5->8
and 13->16 are not working and string is not formatted well. Attached a
picture
. Any further support is highly appreciated.
Cheers
Manuel

6010110254, 2020-08-19:
i'm just like you, the text isn't as clear as some lines don't.

6010110254, 2020-08-19:
[image]

6010110254, 2020-08-19:
[image]
thank you complete

daveythacher, 2020-08-19:
This can be closed.

daveythacher, 2020-08-19:
@manuel1507 How many address pins do you have?

manuel1507, 2020-08-19:
ABC

daveythacher, 2020-08-19:
@manuel1507 What command?

manuel1507, 2020-08-19:
Not sure I got your question..I was trying both demo and python sample.
Result is the same. Maybe some soldering issue...I get two black lines

manuel1507, 2020-08-20:
ok, it has to be some soldering issue. I removed now the HAT and tried the
regular wiring and I managed to get rid out of the black lines, but the
final result is far from being perfect.
See LINK https://youtu.be/jgTPr2ocd_U

If I remove all GND it gets slightly better, but this is not a solution.

Any hint?

This is the command: ./demo --led-rows=16 --led-cols=32 --led-chain=1
--led-parallel=1 --led-show-refresh --led-slowdown-gpio=2 --led-pwm-bits=3
--led-pwm-lsb-nanoseconds=300 -D1 runtext16.ppm --led-gpio-mapping=regular

daveythacher, 2020-08-20:
Lets keep with this command:
./demo --led-rows=16 --led-cols=32 --led-chain=1
--led-parallel=1 --led-show-refresh --led-slowdown-gpio=2 --led-pwm-bits=3
--led-pwm-lsb-nanoseconds=300 -D1 runtext16.ppm --led-gpio-mapping=adafruit-hat

Lets add this to it now:
--led-row-addr-type=3

If this turns on all the display rows that will be progress. At this point we are just trying to explain why some rows are not turning on. It could use a weird multiplex scheme which we can probably work at with different command.

Do you have a picture of the panel back side? Or a list of IC numbers used on your board?

daveythacher, 2020-08-20:
> If I remove all GND it gets slightly better, but this is not a solution.

That is really strange. We can try slowing it down, but I am not sure exactly what that gets. However if you are using certain wire lengths at high speed or in noisy environment it could help. Most of the panels buffer the input signals, so long as the wires are short it should not matter a lot. These panels likely have 4 grounds. Ground loops could be a problem.

Ways to slow it down. Increase one or more of the following --led-slowdown-gpio, --led-pwm-bits, --led-pwm-lsb-nanoseconds (not very effective). You could also increase the columns and maybe mess around with the rows. (But I am not sure I recommend messing with these right now.)

Moving to --led-slowdown-gpio=4 and --led-pwm-bits=8 will seriously drag down the FPS. To drag the multiplex you could try increasing the pwm-bits to 11.

manuel1507, 2020-08-20:
Here the picture

daveythacher, 2020-08-20:
There is no picture

daveythacher, 2020-08-20:
I am not able to see the chips in that picture. Have you tried increasing --led-pwm-lsb-nanoseconds? Did you follow the hook up in here: https://github.com/hzeller/rpi-rgb-led-matrix/blob/90a890ab440f70e85ba7da9c6085c97c3fd67fd2/lib/hardware-mapping.c#L30?

daveythacher, 2020-08-20:
I am curious what is on Q1 and Q2. I think I know what is on U1 and U2. I would guess both are the same. I am not really interested with UR1-4, UG1-4, UB1-4. I think I know that those are, but it could be helpful to know just to be sure. They should all be the same also.

daveythacher, 2020-08-20:
In a deleted comment I mentioned changing the address type. I also mentioned ground loops due to your ground comment. However I am not sure if those made sense per say. However I would be interested in tracking down your panel then checking wiring.

I find it hard to believe that board has soldering issue. You would be able to tell with visual inspection more than likely or with volt meter. You only soldered on the through hole connectors, which should not cause too much issue.

manuel1507, 2020-08-20:
> In a deleted comment I mentioned changing the address type. I also mentioned ground loops due to your ground comment. However I am not sure if those made sense per say. However I would be interested [...]

manuel1507, 2020-08-20:
Hi David,
just to make clear. I was following two paths:
1) RPi4 + Adafruit HAT
2) RPi + breadboard wiring schema and "regular" mapping

Different problems both ways:
1) ./demo --led-rows=16 --led-cols=32 --led-chain=1 --led-parallel=1 --led-show-refresh --led-slowdown-gpio=2 --led-pwm-bits=3 --led-pwm-lsb-nanoseconds=300 -D4

BLACK ROWS (tried already addr option and slow down; I even played around with multiplexing):

[image]

2) Looks much better but it's FLICKERING:

https://youtu.be/BwoGrMfBKcg

I believed I managed to read the chip: Q1 and Q2 are ICN2012. All the others are shift registers from ChipOne.

manuel1507, 2020-08-28:
Habemus Led control. 

[image]
[image]

A little re-wiring and different power adapter and voila.

Thank you David for the all the support and patience. We can close this one.

closed 2020-08-28 by manuel1507

daveythacher, 2020-08-28:
What changes did you make?

manuel1507, 2020-08-28:
When I was using the wires and the breadboard the problem was the power
supply. I put the capacitor in parallel and that solved the issue.

With the Adafruit HAT the issue were two pins not soldered correctly on the
flat cable. Solved.

Thanks again for all your support, I will now play around with the library
and a few applications I had in mind.

Cheers