hzeller/rpi-rgb-led-matrix #1125: 74HC245 works with 3.3V input from pi. Why is it not recommended?

arahasya, 2020-08-24:
I have gone through datasheets of 74hc245 from various manufacturers and found that the voltage range is from 2V to 6V for most .
So I am confused why you have recommended 74HCT245.

https://www.ti.com/lit/ds/symlink/sn54hc245-sp.pdf?ts=1598264859083&ref_url=https%253A%252F%252Fwww.google.com%252F

hzeller, 2020-08-24:
If you operate the HC line with 5V, then the input is requiring at least 3.15V, which _not_ compatible with 3.3V logic level, which is only guaranteed to be 2.4V for high or so.

This is why the HCT line exists: because it is compatible with the TTL input range, which is the guaranteed output range of 3.3V logic.

closed 2020-08-24 by hzeller

arahasya, 2020-08-26:
Okay thanks

If I am manufacturing the boards using HCT is there any reason I still should use active adapter ?
Why isn't the passive adapter recommended for longer chains? It is still protected by the HCT from the first board in the chain.

hzeller, 2020-08-26:
If your input stage of the matrix is HCT, then you technically don't need that on the Pi side. However, you still don't want to drive a longer cable from the Pi to the first panel - the signals are directly generated from the delicate processor core, and I'd be reluctant to drive cable loads with it.