hzeller/rpi-rgb-led-matrix #1126: P5 panel can't turn on

DarkFighterLuke, 2020-08-24:
I bought this 64x32 rgb panel: https://www.amazon.it/gp/product/B06XNJZN89/ref=ppx_yo_dt_b_asin_image_o02_s00?ie=UTF8&psc=1

It doesn't turn on, I followed the wiring of this guide based on this library to connect the display to Raspberry Pi Zero W: https://learn.adafruit.com/connecting-a-16x32-rgb-led-matrix-panel-to-a-raspberry-pi/wiring-the-display

I can see leds turning on only when I disconnect the power supply or the raspberry pi.

Can someone help me?

Thanks a lot.

hzeller, 2020-08-24:
If you use this project, use the wiring guide in this project.
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md

DarkFighterLuke, 2020-08-24:
Ok problem solved, that was an outdated guide with a wrong pinout. 
Following the wiring section of this project solved my problem.

Great job @hzeller , I love this library and how you explained that.

DarkFighterLuke, 2020-08-24:
> If you use this project, use the wiring guide in this project.
> https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md
When  I was writing the post your answer arrived ahahah

DarkFighterLuke, 2020-08-25:
At the first moment I thought that it was normal, but now I've seen that I get a thing like that in the video: https://streamable.com/1e68j7

Is it due to power supply? (I noticed that even when I disconnect power supply I get pretty the same, so maybe the problem is it)

closed 2020-08-25 by DarkFighterLuke