hzeller/rpi-rgb-led-matrix #1127: P5 panel strange behaviour

DarkFighterLuke, 2020-08-25:
At the first moment I thought that it was normal, but now I've seen that I get a thing like that in the video: https://streamable.com/1e68j7

Is it due to power supply? (I noticed that even when I disconnect power supply I get pretty the same, so maybe the problem is it)

daveythacher, 2020-08-28:
Is the power supply on? Is the power supply strong enough to power the display? It is possible the power supply is shutting down due to over current or thermal protection.

You can disable the display with OE. This will turn off the LEDs if the display is browning out. Each panel will be using maybe a couple hundred mA at most. At this point you can try to clear the shift registers to avoid a really large load on power supply.

I have seen stuff like this before. I think a little power leaks back from the HUB75 in some weird way. A lot of the logic internally is probably floating. If multiple rows are selected, which is not supposed to happen, then multiple rows will light. The red LEDs have the lowest turn on voltage. Very little current is likely flowing into each LED.

If the power supply is on and is strong enough. Check to see if the voltages you are sending are valid. However most of those ICs  likely have inputs which will prevent this. However this only works if the main power rail to the panel is within spec. These are designed for 5V, however many will work at 3.3V. 

Adding capacitance may help smooth out high ripple currents. Depending on the display size and panel count you may want to use several with different sizes. This would be an issue if you are working at or near the power limit. Or in cases where the average current is lower than the peak current. Note you may want to use a little care possibly limit the cap inrush to avoid brown out.

Point about using less than 5V:
The color correction however is set for 5V so the white will be incorrect. These uses constant current limiters in the shift registers. This is set via resistor for reference current/voltage which is then used to control the internal drivers. LEDs are diodes so current is exponentially related to voltage. (Called diode IV plot.) 

This means that the resistors set the max current for each RGB color with its own resistor. Each RGB color LED has different forward voltages to reach 20mA on the IV plot. If you lower the expected input voltage the current will be lower. However the current will be further lowered by the IV plot itself.

Newer drivers have programmable logic to solve this. However these are not supported here by this project to my knowledge. I would not recommend using much more than 5 volts. The drivers will prevent excess current, but will heat up a little more.

daveythacher, 2020-08-28:
Can you list the parts on the back of the panel? It can also be caused by an incorrect panel setup or incompatible panel. (Maybe)

DarkFighterLuke, 2020-08-30:
Thank you for the accurate answer! So I think that problem should be the power supply, the voltage is correct but I don't know about the current emitted.

> Can you list the parts on the back of the panel? It can also be caused by an incorrect panel setup or incompatible panel. (Maybe)

Yeah, sure. There are a lot of TC5020A, except for two 74HC245C.

daveythacher, 2020-08-30:
> Yeah, sure. There are a lot of TC5020A, except for two 74HC245C.

There should be another IC type or two on there. However these panels should be compatible.