hzeller/rpi-rgb-led-matrix #1128: P2.5 64x128 panel multiplexing?

Crispdust, 2020-08-25:
Hi all.  I've had the code working with a 64 x 64 panel working using the FM6216A chips.
I've now tried a 64x128 panel and you d that it doesn't look as expected.  Demo D0 and D look like the attached images.
[image]
[image]
These images are with default settings. I've tried the different multiplex settings. And get very different outputs that are all over the place.
Any advice is greatly appreciated.

Dan.

hzeller, 2020-08-25:
did you go systematically through all `--led-row-addr-type` settings and `--led-multiplexing` settings ? Which look best ?

Crispdust, 2020-08-25:
Hi, I went through all the row address settings and all the matrix settings and found the defaults to be best. Other settings seemed to make the image more fragmented or blank, However I didn't try all the various combinations. I'll look at those tonight.  One thing I'll add is that the numbers on the chips are different to what I've seen before.

Crispdust, 2020-08-27:
So any --led-row-addr-type other than 0 results I a blank display no matter what the multiplexing is set to.  
With --led-multiplexing=0 gives the results in the above images.  All the others are more broken up or offset.  9 and 10 cause errors.  10 will display something but 9 leaves the screen blank.
The display uses the hub75E connector and the pinout appears to be the same as the 64 x 64 display that does work.  I've also used the same cables on both displays to rule out a problem with the cables and I've tried a second 64 x 128 display and had the same results.  If you have any further advice it would be much appreciated as it appears to be so close to working correctly.

Crispdust, 2020-08-28:
Hold on, looking at git I believe there are more multiplex options as the software has been updated.  Doh.  I'll check them out as well...

Crispdust, 2020-08-28:
And done it.  I downloaded the latest files off git and found that --led-row-addr-type=4 --led-multiplexing=0 gave me a much brighter output and was close but glitchy. --led-slowdown-gpio=2 sorted out the glitchyness and got it working.  
Apologies for being a pain.  I'd started with the adafruit files a few weeks ago and went from there.

Crispdust, 2020-08-28:
Oh awesome software by the way. You are a star!

closed 2020-08-28 by Crispdust

hzeller, 2020-08-28:
Glad you got it working! Have fun with your LED panels.