hzeller/rpi-rgb-led-matrix #1129: [Question] FM6126B chip | Joy-It RGB-MatrixCtrl

Snoopy913, 2020-08-28:
Hello, 
I got a LED Matrix from ebay and found, that it has the FM6126B chip on it. It is not mentioned here, but does anyone have experience with it and what settings I have to use?

I also found a good looking LED-Matrix Controller from joy-it, after reading the wiring (maybe only available in germany):
[Manual]( https://joy-it.net/files/files/Produkte/RB-MatrixCtrl/RB-MatrixCtrl-Manual.pdf)
[not very detailed datasheet](https://joy-it.net/files/files/Produkte/RB-MatrixCtrl/RB-MatrixCtrl%20Datasheet.pdf )

I than recognised it is refering to your projecthere ;). It does not mention a different mapping setting, so I tried it with different settings.

I received a broken board with a detached condensator, I hope I can fix it somehow, because my soldering kit is stored in a moving box:
[image]

Is it known what this results in? I hoped I can at least try to find the correct settings with this broken board. 
The classic mappping results in this: 
`pi@raspi4B:~/rpi-nsmon/rpi-rgb-led-matrix $ sudo examples-api-use/demo -D 0 --led-rows=32 --led-cols=64 --led-show-refresh --led-gpio-mapping=classic --led-slowdown-gpio=4`

[image]

The regular mappping results in this: 
`pi@raspi4B:~/rpi-nsmon/rpi-rgb-led-matrix $ sudo examples-api-use/demo -D 0 --led-rows=32 --led-cols=64 --led-show-refresh --led-gpio-mapping=regular --led-slowdown-gpio=4`

[image]

The runtext example looks like this: 
`sudo examples-api-use/demo -D 1 ./examples-api-use/runtext.ppm --led-rows=32 --led-cols=64 --led-show-refresh --led-gpio-mapping=classic --led-slowdown-gpio=4`

`https://www.dropbox.com/s/83ppkjid1qxdd8g/VID_20200828_095559.mp4?dl=0`

I hope maybe someone has some experience with this. Thank you

hzeller, 2020-08-28:
If you have a 6126 chip, did you set the `--led-panel-type` accordingly ?
Please read the https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/README.md as it contains all kinds of settings necessary for different types of panels.
The wiring document https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md allows you to compare if it matches your wiring.

Snoopy913, 2020-08-28:
> If you have a 6126 chip, did you set the `--led-panel-type` accordingly ?

Is 6126A and 6126B technically the same? I tried the two different possibilites, but that is now irrelevant.

I just connected the LED matrix without the joy-it HAT and it is running with default settings. And the detached condensator doesn't seem to be a problem.

I guess now I only have to find out which hardware mapping fits to this HAT ´, or create a new mapping.

hzeller, 2020-08-28:
There is a fold-out table for existing alternative mappings at the bottom of this paragraph:
  https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#alternative-hardware-mappings

If you need a new mapping, it is as simple as adding it to [lib/hardware-mapping.c](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/hardware-mapping.c)

Snoopy913, 2020-08-28:
Ok, now everything is working

closed 2020-08-28 by Snoopy913