hzeller/rpi-rgb-led-matrix #1130: Offset only 1 chain out of 3?

multimarcus, 2020-08-31:
Dear Zeller,
Thank you for writing this amazing library and thanks to everyone for adding to it.

This might be a bit of silly question, I am new to C+ and I am learning bit by bit as I go.
Is there a function/way to offset the x or y coordinate of just one chain out of three?
I am working on a custom shape screen and would like to achieve the following, while still retaining a complete image?
Yep, I can edit the image/animation in Photoshop/After Effects, but it would be much quicker to do it through code.

Thanks a lot for your help!
[image]

daveythacher, 2020-09-01:
This can be done is software. I think you need to two mapper instances, but only one of them actually maps to this library. The top level one holds a huge area, while the back one holds the actual display. This would allow you to have dead space in the top one. You could use a third one to drop the dead space, if desired.

The issue with this is it will likely eat a decent amount of memory. However this is not really a concern for the Pi. However there more likely issue will be potentially memory performance and/or the CPU cycles required for this.

There is another way to solve this which is to just install dummy shift registers. This will create the shift. Then a single mapper will have the full display with dead space. This will reduce the amount of memory and CPU cycles. However will put more pressure on the HUB75 performance.

It may end up being a wash. Considering most of this library uses bit banging and the refresh rate is sensitive to the CPU usage.

daveythacher, 2020-09-01:
Note using a FPGA is probably cheaper. You would just internally build a shift registers or delay/counter function of some kind. It would be smaller too.

daveythacher, 2020-09-01:
Is this related to #974 ?

hzeller, 2020-09-01:
You might need to custom-implement a PixelMapper for this https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/pixel-mapper.h . This should be a fairly simple implementation; here some rough pseudocode of the meat:

```c++
include "pixel-mapper.h"

class ShiftMapper : public rgb_matrix::PixelMapper {
  virtual const char *GetName() const { return "ThirdShiftMapper"; }
  virtual bool GetSizeMapping(int matrix_width, int matrix_height,
                              int *visible_width, int *visible_height) const {
    *visible_height = matrix_height;
    *visible_width = matrix_width + 32;   // because our square we cover is one panel wider
    return true;
  }

  void MapVisibleToMatrix(int matrix_width, int matrix_height,
                          int visible_x, int visible_y,
                          int *matrix_x, int *matrix_y) const {
    *matrix_y = visible_y;  // y always stays the same
    *matrix_x = visible_x;
    if (visible_y > 64)  { // third chain
      *matrix_x += 32;   // or minus, depending on shift
    }
  }
};
```

Then somewhere before you call `RGBMatrix::CreateFromFlags()`, register it:

```c++
rgb_matrix::RegisterPixelMapper(new ShiftMapper());
```

You then can use your new mapper by specifying on the command line `--led-pixel-mapper=ThirdShiftMapper`

If you have trouble with the c++, ask someone you know who does; they will be able to make something out of the above.

daveythacher, 2020-09-01:
This is an example of saving memory by murdering your CPU! I was suggesting memcpy.

hzeller, 2020-09-01:
? where would you murder the CPU ?

daveythacher, 2020-09-01:
It works off scanning every pixel, but maybe only once per frame. It appears to be a linear copy, so it may not even be saving memory.

I am not sure about the cache performance or the cycle usage.

hzeller, 2020-09-01:
The mapper is applied once at initialization time; nothing is used at runtime. And the code pushing data to the panels is only concerned about the actual panels, so there is now overhead in memory or runtime.

(David, I'd like you to please only respond to user issues if you know the details. You have helped users with the correct information before, so please more of that. But for things you can only speculate it is probably better to not add confusing information).

daveythacher, 2020-09-01:
> The mapper is applied once at initialization time; nothing is used at runtime. And the code pushing data to the panels is only concerned about the actual panels, so there is now overhead in memory or runtime.

So if I change the frame this function never gets called? If am not mistaken this function converts Canvas frames into local storage of framebuffer, which is looped over. User draws canvas then this is converted via pixel mapper to matrix format.

hzeller, 2020-09-01:
Yes you are mistaken. The mapper is used to set-up a look-up table initially, then never used again.

daveythacher, 2020-09-01:
> Yes you are mistaken. The mapper is used to set-up a look-up table initially, then never used again.

This function is a lookup. The lookup still exists/is used with or without it. (setPixel) I think my points are valid based on the look up.

multimarcus, 2020-09-02:
Hey! Thanks a lot for all of the above! I will have a go at it over the next few days and will send an update!!!