hzeller/rpi-rgb-led-matrix #1132: Cannot make video-viewer

vic-cortes, 2020-09-03:
Im trying to `sudo make video-video` but have issues. The output is as follows:

g++ -I../include -O3 -W -Wall -Wextra -Wno-unused-parameter -D_FILE_OFFSET_BITS=64 -std=c++11 -c -o video-viewer.o video-viewer.cc
video-viewer.cc: In function 'int main(int, char**)':
video-viewer.cc:282:7: error: 'AVCodecParameters' was not declared in this scope
       AVCodecParameters *codec_parameters = nullptr;
       ^
video-viewer.cc:282:26: error: 'codec_parameters' was not declared in this scope
       AVCodecParameters *codec_parameters = nullptr;
                          ^
video-viewer.cc:285:56: error: 'AVStream' has no member named 'codecpar'
         codec_parameters = format_context->streams[i]->codecpar;
                                                        ^
video-viewer.cc:298:76: error: 'av_guess_frame_rate' was not declared in this scope
       AVRational rate = av_guess_frame_rate(format_context, stream, nullptr);
                                                                            ^
video-viewer.cc:303:72: error: 'avcodec_parameters_to_context' was not declared in this scope
       if (avcodec_parameters_to_context(codec_context, codec_parameters) < 0)
                                                                        ^
video-viewer.cc:354:42: error: 'av_packet_alloc' was not declared in this scope
       AVPacket *packet = av_packet_alloc();
                                          ^
video-viewer.cc:373:58: error: 'avcodec_send_packet' was not declared in this scope
             if (avcodec_send_packet(codec_context, packet) < 0)
                                                          ^
video-viewer.cc:376:66: error: 'avcodec_receive_frame' was not declared in this scope
             if (avcodec_receive_frame(codec_context, decode_frame) < 0)
                                                                  ^
video-viewer.cc:405:29: error: 'av_packet_free' was not declared in this scope
       av_packet_free(&packet);
                             ^
Makefile:40: recipe for target 'video-viewer.o' failed
make: *** [video-viewer.o] Error 1

I had a similar issue with `sudo make text-scroller` but adding `-std=c++11` in `CXXFLAGS` solve the [issue](https://github.com/facebookresearch/faiss/issues/352).

However for `video-viewer` had no success. I've already tried uninstalling & installing the dependencies.

I'm using a Rpi3 with the following image:
`
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
`
**Kernel info**: `Linux raspberrypi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux`

Any suggestion?

hzeller, 2020-09-03:
The libraries mentioned in https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/utils#building-2 you installed ?

hzeller, 2020-09-03:
Jessie is a pretty ancient version of Raspbian; it might have too old libav libraries.
Try a [current one](https://www.raspberrypi.org/downloads/raspberry-pi-os/) or [DietPi](https://dietpi.com/).

vic-cortes, 2020-09-03:
Ok, I'll switch to newer version and let you know. Thanks

rijotech, 2020-09-19:
try updating your distro libraries first.
sudo apt-get update
sudo apt-get upgrade 
sudo make video-video

closed 2025-11-27 by marcmerlin