hzeller/rpi-rgb-led-matrix #1133: RPi 4 unable to handle more than 2 64x64 panels on a chain

PHimmel, 2020-09-04:
Hello Mr. Zeller. Thank you so much for your contribution with this software, as well as for actively maintaining it and resolving issues posted here. 

I have had success using a RPi 3 B+ for running 12 64x64 panels flawlessly. After a hapless short circuit involving its GPIOs, I obtained a RPi 4 B 2GB. I had anticipated it would outperform the 3 but so far have hit a wall.

[image]

My current configuration arguments are: --led-rows=64 --led-cols=192 --led-parallel=3 --led-pixel-mapper=V-mapper
I have tried a multitude of other combinations as well - all ultimately resulting with the same issue. I have the --led-slowdown-gpio set all the way to 5. (I notice that when I put it on 4, the almost identical issue emerges on the following panel in the chain[is the RPi 4 really just too fast?]) 

Any panel more than 2 results in the same error in the photo. The issue manifests within a consist period of time (generally every ~2 seconds), and always involves the lower 32 rows and the rightmost 16 columns --- which extend to the top.

It is also exacerbated by brightness (increased brightness == increased faulty display).

I have checked my wiring repeatedly as well as fully rewired multiple times too.
I got a fresh microsd card and tried Raspian Lite, DietPi and regular Raspian - all with the same issue.

I tried every additional flag and received good success with --led-pwm-lsb-nanoseconds=~50 as well as with --led-pwm-dither-bits=1. Interestingly, --led-no-pulse-hardware helps a bit as well. . . It's just that the issue is still present and it feels as though it is only masking the underlying root cause of the problem. - Also, using those modifications greatly increases CPU usage(I suppose it is artificially slowing it down, which in-and-of itself adds to the CPU load?). I also reserved the 4th core for refreshing, as you recommend.

[image]

I'm using a 5V80A SMPS. It is outputting ~5.07V and at the inlet it is around ~5.03V. I added only one 4700 µf so far, to a test panel to see what affect it had on the issue - none.

[image]

This is the level-shifting circuit that I made. I actually rebuild/rewired it in hopes that it would clear up the problem. I followed your basic format in the active-shifting board (save the nice interfaces and exact ordering of the lines).

My total setup ended up working superbly (after some mild tweaking) and I was learning the library to create my own programs and never expected this to occur upon the simple change of one pi to another (faster) one.

Thank you again Mr. Zeller, I have spent many hours learning and working with the programs that you have made and following the instructions that you created. It has been a challenging, fulfilling experience that has resulted in a beautiful display.

PHimmel, 2020-09-04:
Update:

Apparently the constraints for --led-slowdown-gpio are relatively arbitrary and can be adjusted in led-matrix.cc, here:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/1ca622655a8573fffd646d2ea24fb2fdc85d70ae/lib/led-matrix.cc#L626
(After making the adjustment be sure to recompile, aka re-'make', the git clone)

Using a slowdown value of 6 has pretty much taken care of the aforementioned issue! (There still is a bit of vertical ghosting present when on a black background).

If the upper-limit of the cap is arbitrary, and won't cause any detrimental effects, then why not raise it higher than 5? I'd imagine this would be a relatively common problem, although I haven't seen much mention of it in Issues.

daveythacher, 2020-09-05:
Hzeller did some testing a while back and found the GPIO performance of the Pi's differed quite a bit. The Pi 4 was supposedly capable of 131 MHz of performance. This will can cause all kinds of issues. He also uses BCM, which creates a few potential issues along this line.

Fundamentally this project is based in bit banging with a weak notion of time. Many of the panels supported by this project have a recommended clock speed of 25MHz. The drivers and wire also have frequency limits.  What this project does is throw a full core/thread at the problem. The stronger the core the crazier this problem becomes. (As you saw. The Pi 3 has something like 58MHz.)

Normally you would use a piece of hardware, interrupts, DMA, etc. to throttle the transfers correctly. Since the Pi was designed to be a cheap computer for teaching programming it follows a more micro-processor/generic computer design. (Not a micro-controller/real time system.) That being said it could be interfaced with hardware to make up for this, however low cost was apparently a major consideration.

This means the software directly toggles the pins. The Pi supports a single GPIO bank which makes things a little cleaner internally. However doing square wave PWM is problematic for computation. It uses a ton of cycle, so he does BCM. This is done using the OE signal driven by one of the two PWM generators. For the serial data he bit bangs, which naturally is problematic. 

Hence the gpio-slowdown, which is arbitrary. This attempts to get the clock down, but can have some issues. Again the expectation is the he has a full core. So he will deadlock it. Really weird stuff can happen when the system starts load balancing. This is documented.

Much of the interface timing notions are arbitrary. In part since the bit bang operations are shared with the computation cycles. Thus performance can vary depending what the system is doing. This can be used to some extent as a natural slowdown. However the matrix is dumped in a single routine. So changing the GPIO slowdown can help.

The OE times (pwm-lsb-nanoseconds) are interesting if you want to look into the code. In particular since this setting may not correctly correspond with the OE hold times or data transfer expectation times for the chain lengths. You just fiddle with a few settings till it looks right, which may work correctly some times and wrong at others. 

Temperature for example can throw things off. Datasheets list the recommend for the parts. Some will support better performance while others will not. So if you setup for a good part then it will not work for one of the bad ones. However the bad one will still comply with the recommendations of the datasheet. Over clockers refer to this as the silicon lottery.

12 64x64 RGB matrixes in a single chain is quite a lot pixels to service under 25MHz. It can be done however the number color steps will be less. This library will potentially report higher PWM bit values than what is actually happening. It will clip the lower bits, but this is somewhat of a speculation. I leave that to your own judgement.

One thing that seems to be common for software driven panels is the failure to understand certain notions of period or bandwidth. I will not pretend to be the most knowledgeable of AC or analog systems. However magic is how I would classify a number of claims from these panel driver implementations. 

Maybe I am wrong, but no one has told me in a manner I should take seriously. I once had PWM criticized as "naive" which I found funny. Why is in here.

Another area of concern is the power supply age, power output, operational status, etc. Capacitance can be an issue with these too. Again since he uses such high frequencies he can have low pass filters. He can also have high inrush requirements, which will act like a low pass filter also.

What pwm-bit count where you using with the Pi 3? Are you able to use it with the Pi 4? I am curious.

If you are convince the GPIO-slowdown is a fix then close this issue. File another issue requesting an enhancement to the GPIO slowdown for the Pi 4.