hzeller/rpi-rgb-led-matrix #1134: Segmentation fault

mushfiq-heraclius-xen, 2020-09-05:
@hzeller  i have put all the codes of scrolling text inside a function and did all the necessary changes , but i'm getting Segmentation fault whenever i try to call that function from the main, inside main i'm just running a simple socket server, can u please tell me why is this happening ?

here is my code for your convenience 

void scrollTextonRGBLEDMatrix(std::string line)
{
	RGBMatrix::Options matrix_options;

	matrix_options.hardware_mapping = "regular";  
	matrix_options.rows = 32;
	matrix_options.cols = 64;
	matrix_options.chain_length = 2;

	rgb_matrix::RuntimeOptions runtime_options;
	// If you drop privileges, the root user you start the program with
	// to be able to initialize the hardware will be switched to an unprivileged
	// user to minimize a potential security attack surface.
	//runtime_defaults.drop_privileges = 1;
	runtime_options.gpio_slowdown = 4; 
	
	rgb_matrix::ParseOptionsFromFlags(nullptr, nullptr, &matrix_options, &runtime_options);

	/* x_origin is set by default just right of the screen */
	const int x_default_start = (matrix_options.chain_length* matrix_options.cols) + 5;
	int x_orig = x_default_start;
	int y_orig = 0;
	int letter_spacing = 0;
	float speed = 7.0f;
	int loops = -1;
	
	const char *bdf_font_file = "../fonts/10x20.bdf";
	rgb_matrix::Font font;
	font.LoadFont(bdf_font_file);
	Color color(255, 0, 0);
	Color bg_color(0, 0, 15);

	

	RGBMatrix *canvas = RGBMatrix::CreateFromOptions(matrix_options, runtime_options);
	//if (canvas == NULL)
	//	return 1;
	
	const bool all_extreme_colors = (matrix_options.brightness == 100) && FullSaturation(color) && FullSaturation(bg_color);
	if (all_extreme_colors)
		canvas->SetPWMBits(1);
	
	// Create a new canvas to be used with led_matrix_swap_on_vsync
	FrameCanvas *offscreen_canvas = canvas->CreateFrameCanvas();
	int delay_speed_usec = 1000000;
	if (speed > 0) 
	{
		delay_speed_usec = 1000000 / speed / font.CharacterWidth('W');
	} 
	else if (x_orig == x_default_start) 
	{
		// There would be no scrolling, so text would never appear. Move to front.
		x_orig = 0;
	}

	int x = x_orig;
	int y = y_orig;
	int length = 0;

	while (loops != 0) 
	{
		offscreen_canvas->Fill(bg_color.r, bg_color.g, bg_color.b);
		// length = holds how many pixels our text takes up
		length = rgb_matrix::DrawText(offscreen_canvas, font, x, y + font.baseline(), color, nullptr, line.c_str(), letter_spacing);

		if (speed > 0 && --x + length < 0) 
		{
			x = x_orig;
			if (loops > 0) 
			--loops;
		}
		// Swap the offscreen_canvas with canvas on vsync, avoids flickering
		offscreen_canvas = canvas->SwapOnVSync(offscreen_canvas);
		usleep(delay_speed_usec);
	}
	// Finished. Shut down the RGB matrix.
	delete canvas;
}

hzeller, 2020-09-05:
Do you intend to start and stop the matrix for evrery invocation of that function ? Usually, you'd create the matrix once and pass the pointer to the function that fills it.

Why do you call ParseOptionsFromFlags() when you don't pass the flags ?

mushfiq-heraclius-xen, 2020-09-05:
problem solved @hzeller thank you very much i ommited that line :) <3

mushfiq-heraclius-xen, 2020-09-05:
@hzeller i'm getting compiler error 
 ‘matrix_options1’ does not name a type
 matrix_options1.hardware_mapping = "regular";
whenver i try to declare the following lines as global variable 
RGBMatrix::Options matrix_options1;
matrix_options1.hardware_mapping = "regular";  
what is the correct way to declaring global variable as this settings are constant

hzeller, 2020-09-05:
You'd typically declare it global and then assign things in main:

```c++
RGBMatrix::Options global_options;

int main(int argc, char **argv) {
   global_options.hardware_mapping = "regular";
   // ...
}
```

daveythacher, 2020-09-05:
Try:
struct RGBMatrix::Options matrix_options;

The compiler may have optimized. C++ treats this a little different. However I am not sure this is technically a type, which I think is bad for the library.

You can declare it within main. You can localize it with static either inside main or global to main's global scope. Rather than allowing it to be everywhere.

Overall I think what you are doing with scrollTextonRGBLEDMatrix will work, but is not exactly ideal. Ideally you would configure and start the main RGBMatrix/original canvas from main. Or some other broader execution thread that will live longer than a single function call. 

RGBMatrix spins up a thread in the background, allowing yours to draw the offscreen. When you delete the RGBMatrix the panels is not longer driven. I see no point to do this. You could blank the active canvas.

scrollTextonRGBLEDMatrix can create the offscreen buffer. I think double buffering is the right approach here. What would be removed is the offscreen buffer at the end of the function. The main RGBMatrix/active canvas gets removed by the end of the main or the broader execution thread responsible for the display.